openai/roboschool #2: opengl_init failed

suriyadeepan, 2017-05-16:
Hi guys!

I'm getting this error when i run
`python3 $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py`

```bash
[2017-05-16 12:14:12,580] Making new env: RoboschoolHumanoid-v0
[2017-05-16 12:14:12,582] Making new env: RoboschoolHopper-v0
[2017-05-16 12:14:12,582] Making new env: RoboschoolWalker2d-v0
python3: render-simple.cpp:676: void opengl_init(const boost::shared_ptr<Household::World>&): Assertion `fmt1.majorVersion() == fmt2.majorVersion()' failed.
Aborted (core dumped)
```
Can you help?

olegklimov, 2017-05-16:
I was afraid of this :(  Yes you can help, please do. What are your video drivers, what fmt1 and fmt2 report, does it work if you remove the assert?

DeepakParamkusam, 2017-05-16:
Same error while running `demo_race2.py`. I am running Ubuntu 16.04 with Intel HD Graphics, i915 kernel driver. 
My fmt report -
 
<pre>fmt1.minorVersion() = 1
fmt1.majorVersion() = 4
fmt2.minorVersion() = 3
fmt2.majorVersion() = 3</pre>

Removing the assert (both majorVersion and minorVersion) does not work. Gives QGLShader error.

<pre>[2017-05-16 16:32:01,238] Making new env: RoboschoolHopper-v0
[2017-05-16 16:32:01,241] Making new env: RoboschoolHumanoid-v0
[2017-05-16 16:32:01,243] Making new env: RoboschoolHumanoid-v0
QGLShader::compile(Vertex): 0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES

QGLShader::compile(Fragment): 0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES

simple_texturing.frag.glsl LOG: '0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES
'
QGLShader::link: "error: no shaders attached to the program\n"
python3: render-simple.cpp:243: void SimpleRender::Context::initGL(): Assertion `r0' failed.
Aborted (core dumped)
</pre>

olegklimov, 2017-05-16:
Intel HD graphics, that is out in 2011? It it supposed to support OpenGL 3.3, according to this table:

https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_units

I'll try to downgrade shaders to 3.3, if it's easy to do, sure why not (In a few days.)

peter-ch, 2017-05-17:
I have this problem. I think it's related: 

> py agent_zoo/demo_race2.py 

[2017-05-17 23:27:46,388] Making new env: RoboschoolHopper-v0
[2017-05-17 23:27:46,393] Making new env: RoboschoolHalfCheetah-v0
[2017-05-17 23:27:46,394] Making new env: RoboschoolWalker2d-v0
QGLShaderProgram: could not create shader program
bool QGLShaderPrivate::create(): Could not create shader of type 2.
python3: render-simple.cpp:242: void SimpleRender::Context::initGL(): Assertion `r0' failed.
Aborted (core dumped)

duo51, 2017-05-17:
[](url)

gleybo, 2017-05-18:
same issue here as with @peter-ch - 

python3 $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py
[2017-05-18 16:14:22,830] Making new env: RoboschoolWalker2d-v0
[2017-05-18 16:14:22,833] Making new env: RoboschoolHopper-v0
[2017-05-18 16:14:22,834] Making new env: RoboschoolHumanoid-v0
QGLShaderProgram: could not create shader program
bool QGLShaderPrivate::create(): Could not create shader of type 2.
python3: render-simple.cpp:242: void SimpleRender::Context::initGL(): Assertion `r0' failed.
Aborted (core dumped)

I'm running on TitanX, with driver version: 375.39, Ubuntu 16.

Any ideas?

gleybo, 2017-05-18:
Upgrading the Nvidia driver to v375.66 made the errors go away, and the simulator is now starting to run - but now it is only rendering a black screen :(

This is although scores are running on the screen, so I'm guessing that the physics engine is up and running.

peter-ch, 2017-05-18:
I guess we need some less advanced 3D engine.

cjjeakle, 2017-05-19:
I hit the same sequence as @gleybo and have upgraded to Nvidia driver v376.66 (I'm using a GTX970 on Ubuntu 16.04).

#### My current error:
`composeAndFlush: makeCurrent() failed` continually appears in the console, with a rendered window appearing that contains a fixed image of whatever was under where the window initially appeared.

The title bar updates with "0 objects", "bullet" having a non-0 per-frame time (<1 ms, and continually changing), and "render" showing 0 ms per frame. As a consequence, it looks like a combination of rendering issues, and perhaps some failed initialization.

I am going to give some debugging a shot, and will follow-up if I make any progress.

#### Tips for folks still getting everything running:
I had to take these (fairly straightforward in retrospect) steps to get where I am:
* Ensure you install both Gym and Roboschool using pip3.
* Switch to the latest stable driver for your graphics card
(check here for nvidia: https://www.nvidia.com/object/unix.html).

olegklimov, 2017-05-20:
> GTX970 on Ubuntu 16.04
> makeCurrent() failed

Does tux racer work?

If it works, can you please look for "turn on shadows" in Makefile and see if you can get it running without shadows?

olegklimov, 2017-05-20:
I've improved error handling for OpenGL init, and also 3.3 shaders should work, but I have no computer with 3.3 to check. Please test.

DeepakParamkusam, 2017-05-20:
It works for me now. Thank you.

suriyadeepan, 2017-05-22:
works for me too! thanks @olegklimov :+1:

closed 2017-05-22 by suriyadeepan

PabTorre, 2019-01-22:
If you're getting this error using Win 10's WSL Ubuntu and XLaunch. 
Make sure you're not running XLaunch on "Multiple windows mode", but rather in "Single big window mode"

tim-impactia, 2019-06-30:
Using the docker container on Windows 10 and VcXsrv/XLaunch it also worked with WindowMode="Windowed" in the config.xlaunch.

With WindowMode="MultiWindow" I had to turn off native OpenGL by setting Wgl="False" in config.xlaunch. I didn't notice any real difference in performance.

In case it helps anyone else: the general setup is based on [https://dev.to/darksmile92/run-gui-app-in-linux-docker-container-on-windows-host-4kde](url) and is very straightforward:

After installing VcXsrv with Chocalatey and running XLaunch with one large window I started the "roboschool" container with:

docker run -ti --rm -e DISPLAY=192.168.1.143:0.0 roboschool

Where 192.168.1.143 should be replaced with your ip address

HareshMiriyala, 2019-09-09:
Had the same trouble. Using pycharm to run my python script. Just restarting my pycharm instance worked.