openai/roboschool #4: Issue with building

tjacobs, 2017-05-20:
Hi, on my 10.12.4 MacOS system, I'm having trouble building the CPP element.

After installing a few missing requirements with brew, I'm now running into this issue: 
```
Thomass-MacBook-Air:cpp-household thomasjacobs$ make
Mac, assumed python3, homebrew branch
Link against python 3.6
Software render
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers -I/usr/local/Cellar/assimp/3.3.1/include -I/opt/X11/include/GLES3 -I/usr/local/Cellar/boost/1.64.0_1/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
render-simple.cpp:280:2: error: use of undeclared identifier 'glTexStorage2D'
        glTexStorage2D(GL_TEXTURE_2D, 1, GL_RGBA8, W, H);
        ^
render-simple.cpp:284:2: error: use of undeclared identifier 'glTexStorage2D'
        glTexStorage2D(GL_TEXTURE_2D, 1, GL_DEPTH24_STENCIL8, W, H);
        ^
render-simple.cpp:298:2: error: use of undeclared identifier 'glTexStorage2D'
        glTexStorage2D(GL_TEXTURE_2D, 1, GL_RGBA8, W16, H);
        ^
3 errors generated.
make: *** [.build-release/render-simple.o] Error 1
```

I've set:
```
export PATH=/usr/local/bin:/usr/local/opt/qt5/bin:$PATH
export PKG_CONFIG_PATH=/usr/local/opt/qt5/lib/pkgconfig
```
and the brew installs return:
```
Thomass-MacBook-Air:cpp-household thomasjacobs$ brew install python3
Warning: python3-3.6.1 already installed
Thomass-MacBook-Air:cpp-household thomasjacobs$ brew install cmake tinyxml assimp ffmpeg qt
Warning: tinyxml-2.6.2 already installed
Warning: assimp-3.3.1 already installed
Warning: qt is a keg-only and another version is linked to opt.
Use `brew install --force` if you want to install this version
Error: cmake 3.7.2 is already installed
To upgrade to 3.8.1, run `brew upgrade cmake`
Error: ffmpeg 3.3 is already installed
To upgrade to 3.3.1, run `brew upgrade ffmpeg`
Thomass-MacBook-Air:cpp-household thomasjacobs$ brew install boost-python --without-python --with-python3 --build-from-source
Warning: boost-python-1.64.0 already installed
```

I've tried force installing qt and relinking it.
I've tried in an anaconda env as well, to no avail.
What's my next step here?

olegklimov, 2017-05-20:
Great, let's fix it!

Check that you have:

```
/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h
```

header, `glTexStorage2D()` function inside it.

This header is included in `render-simple.cpp` line 8.

And finally the compiler is able to find this header, because the path is explicitly typed into Makefile.

Something in this list isn't working. Can you tell what it is?

(strangely, it seems to include the file, but the function is not there, xcode version?)

tjacobs, 2017-05-21:
Thanks! Looks like I'm missing the headers in my OpenGL.framework. 

```
Thomass-MacBook-Air:cpp-household thomasjacobs$ ls /System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h
ls: /System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h: No such file or directory
Thomass-MacBook-Air:cpp-household thomasjacobs$ ls /System/Library/Frameworks/OpenGL.framework/Headers/
ls: /System/Library/Frameworks/OpenGL.framework/Headers/: No such file or directory
Thomass-MacBook-Air:cpp-household thomasjacobs$ ls /System/Library/Frameworks/OpenGL.framework/
Libraries	OpenGL		Resources	Versions
```

Is Xcode being up to date a dependency?

I'll update it and try reinstalling developer tools and OpenGL framework.

tjacobs, 2017-05-21:
Thanks, got it working now! Chase that ball puny human!

I wrote a little script to help myself and others get it building on mac:
https://github.com/tjacobs/roboschool/commit/9a2bea64604950d6003eb5b08ac514fc37577e9d

[image]

tjacobs, 2017-05-21:
These demos seem mean to the robots, almost as mean as boston dynamics' beating up of real robots https://twitter.com/TomPJacobs/status/866121130217676800

closed 2017-05-22 by tjacobs

rdaries29, 2017-07-21:
Hi,
I see to have the same problem with the missing header file. What did you do to overcome this?
[image]

rdaries29, 2017-07-21:
Hello,

I was able to use your script to install the missing headers directory for OpenGL framework but I seem to be getting an error when I attempt to create the makefile. Anything I could do to resolve this?

[image]

olegklimov, 2017-07-21:
You try to write to filesystem root "/". This is not a bug in Roboschool what you're figuring out. This "issues" section is for bugs in this project.

wsgdrfz, 2018-04-15:
What if I don't have:
`/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h
`
In my OpenGL.framework: 
[image]