openai/roboschool #5: Can I load URDF robots?

tjacobs, 2017-05-22:
How can I load URDF model robots (i.e. ones with STL models as parts), such as Fetch and Atlas, and my own? Is there a demo script for these currently?

tkelestemur, 2017-05-23:
Same here, is it possible to do that?

benelot, 2017-05-24:
Bullet Physics can do this out of the box, but I am not sure if it is accessible through roboschool.

olegklimov, 2017-05-24:
A word of advice. The focus of this project is research. If you want realistic simulation for realistic robots, you're better off with ROS and Gazebo and V-REP. Use this project for simplified environments to study or improve learning methods you use.

That said, there are Atlas and Fetch model in this repository. Atlas works with modifications I have in the works and without self-collision, Fetch doesn't work yet. And no, these models don't work in Bullet and pybullet. It's doable, but a bit more work is required.

For Atlas and Fetch environments to officially appear here, first thing to figure out is what set of tasks will be most useful. Once again, the focus is research.

tjacobs, 2017-05-25:
Thanks Oleg, yep, this is just for some basic simulation of joint movements. I've used Gazebo but it's overkill for what I need and a pain to set up. I'm simulating my robot which is described in URDF format, and it's currently being simulated in Bullet via pybullet fine, I'm just wondering how to bring it over to the nice roboschool UI, and then I can start trying out some walking policies as well.

I'll await your work on the Atlas loading, thanks!

olegklimov, 2017-05-25:
If you know where to modify your own robot, you have good chances to have it working in Roboschool. Compared to pybullet, python binding here is arguably better (objects are objects, not int handles), but still not obvious choice. It will take bigger environments that having C++ part will show advantages.

olegklimov, 2017-05-25:
Let's keep it open for others to see.

olegklimov, 2017-07-18:
We have Atlas in v1.1. Let's close this.

closed 2017-07-18 by olegklimov

tjacobs, 2017-07-19:
Nice! Thanks! I'll try him out.

benelot, 2017-07-19:
How do I get version 1.1?

olegklimov, 2017-07-19:
It's there.

ricardotellez, 2017-07-28:
Hi tjacobs.
I think that this tool can be useful for what you need: [Ros Development Studio](http://rds.theconstructsim.com)
It is a web development environment for ROS including already working simulations. No need to install or configure anything, and you can use any computer to program and test in ROS.

Recently we included OpenAI Gym support. You can check how it works here: https://youtu.be/0sNWqtCGTAQ

Also if you need to learn more about how to use OpenAI with ROS and Gazebo sims I would recommend you this tutorial: https://youtu.be/o9FpZ1QQqcc

Finally, if you wan to know more about URDF robots, try this tutorial: https://youtu.be/UUwHK5ONTAQ

Hope it helps.

meppe, 2018-07-06:
Hi, 
I was trying to port the Hindsight Experience Replay (HER) experiments with the Fetch robot from the mujoco-based simulation of OpenAI baselines to roboschool, but this turned out to be more complicated than expected. I would highly appreciate an integration!

benelot, 2018-07-06:
Since roboschool is starting to age as it is clamped to a certain version
of bullet, I would recommend to integrate it with pybullet, which features
the same mujoco ports as roboschool but is much more actively maintained.
Check out pybullet on github.