openai/roboschool #6: Roboschool renders a black screen

gleybo, 2017-05-23:
Hi,

Following the installation instructions, and updating Nvidia driver to the latest one (as I have reported here: #2) , Roboschool is running but renders only a black screen.

My setup is Ubuntu 16 based, with an Nvidia TitanX. The added fixes for OpenGL init and 3.3 shaders did not help to fix this issue, so I suspect that this is a different issue than in #2.

olegklimov, 2017-05-24:
Hi. Let's fix this as well. I have GTX Titan X (Maxwell) on 16.04 box and it works. Can you please post more info: error messages, any information that can help?

gleybo, 2017-05-25:
Thanks. 

There are no error messages on screen. It seems that from Roboschool's side, it is not aware of any issue. 

~$ python3 roboschool/agent_zoo/RoboschoolHumanoid_v0_2017may.py 
[2017-05-25 11:25:38,894] Making new env: RoboschoolHumanoid-v0
score=3691.15 in 1000 frames
score=3708.30 in 1000 frames

[image]

Are there any logs being dumped which I can grab and attach here?

cjjeakle, 2017-05-25:
Hi All,

I was where @gleybo was for some time and decided to do a clean install of Ubuntu 17 to debug without any of my existing configs. It took some puzzling, but I've got a working environment, and took some detailed notes. I only use `sudo` where explicitly noted below.

### Install:
#### Important notes:
1. This appends `ROBOSCHOOL_PATH` to your `~/.bashrc`
2. This assumes the path `~/Projects/` exists and you run the following in that directory

``` bash
sudo apt install -y python3-pip
pip3 install gym

apt install cmake ffmpeg pkg-config qtbase5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev

git clone https://github.com/openai/roboschool.git
cd roboschool
echo "ROBOSCHOOL_PATH=~/Projects/roboschool" >> ~/.bashrc
source ~/.bashrc

git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision
mkdir bullet3/build
cd    bullet3/build
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..
make -j4
make install
cd ../..

pip3 install -e $ROBOSCHOOL_PATH
```

### Fix loader issues:
#### Important notes:
1. LD_LIBRARY_PATH is overwritten, ensure it's empty before proceeding `echo $LD_LIBRARY_PATH`

The following steps fix a loader issue I hit:
ImportError: libBulletCollision.so.2.87: cannot open shared object file: No such file or directory

```bash
# This must be done every time you open the terminal (put it in .bashrc to automate this)
LD_LIBRARY_PATH=~/Projects/roboschool/roboschool/cpp-household/bullet_local_install/lib
export LD_LIBRARY_PATH
```

### Fix nvidia driver issues:
#### These steps can go horribly wrong. Take backups and only run them if necessary. I've compiled them to save others time, but they may not be exactly what will work for you.

These steps resolved the following error for me:
QGLShaderProgram: could not create shader program
bool QGLShaderPrivate::create(): Could not create shader of type 2.

I hypothesize this will fix the black screen issue, too.

1. Go to https://www.nvidia.com/object/unix.html and download the latest driver for your card
2. Open tty1 (ctrl + alt + F1) and sign in
3. Stop your x server: `sudo service lightdm stop`
4. Remove existing nouveau drivers `sudo apt --purge remove xserver-xorg-video-nouveau`
5. Remove existing nvidia drivers `sudo apt purge nvidia-*`
6. `cd ~/Downloads`
7. `sudo sh ./NVIDIA_DRIVER_NAME.run`
    * I got a warning that the system provided pre-script check failed, but I felt comfortable continuing. YMMV if you see the same situation.
8. Start your x server: `sudo service lightdm start`
9. Log in, open a terminal, set `LD_LIBRARY_PATH`
10. `python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py`
11. Enjoy :)

### SUCCESS:
[image]

gleybo, 2017-05-25:
Thanks a lot @cjjeakle.
Following your steps on "Fix nvidia driver issues" indeed resolved the issue. :-)

closed 2017-05-25 by gleybo

cjjeakle, 2017-05-26:
I'm glad it worked!

It may be worthwhile to re-run `pip3 install -e $ROBOSCHOOL_PATH` now that the drivers are set up. On my machine, dependencies in `$ROBOSCHOOL_PATH/roboschool/cpp-household` were building without SSAO enabled (as the drivers couldn't be located). Now all the features should be enabled :100:.

olegklimov, 2017-05-29:
> LD_LIBRARY_PATH

Hey @cjjeakle thanks for your description, but why do you need `LD_LIBRARY_PATH`  for your setup? It is never used in the project for any purpose. Local Bullet install is referenced to using `-rpath` linker instruction (in Makefile), so you don't have to set any environment variables, except for the linuxbrew case.

olegklimov, 2017-05-29:
Another `LD_LIBRARY_PATH` problem in #10 ...

cjjeakle, 2017-05-30:
I'm unsure. I saw that everything was in order using `readelf`, but it looks like "meta-references" are having issues. Eg: cpp-household has a proper rpath and loads dependencies fine, but the .so files it loads don't search that rpath (or even their own directory) for their own dependencies.

I only hit this on Ubuntu 17, not on Ubuntu 16, so I believe it's a bug in the OS or its a flaw in the default OS configuration. The loader path step is definitely more of a workaround than a fix.