openai/roboschool #7: Reacher crashes after second call to reset

winfried-loetzsch, 2017-05-24:
I am using this code to run the RoboschoolReacher-v0 environment. The first call to env.step in the second iteration of the outer for loop produces a segmentation fault error.

    env = gym.make("RoboschoolReacher-v0")
    for i in range(20):
        obs = env.reset()

        for j in range(100):
            obs, r, done, _ = env.step(np.array([0,0]))
            data = env.unwrapped.render("rgb_array")

I also found that the variables theta_dot and gamma_dot of RoboschoolReacher were set to nan. The reacher example from the agent_zoo produces a similar error: QImage::setPixel: coordinate (1017,-4910411) out of range (occurs multiple times with different values).

When I remove the call to render, everything works fine.
Am I getting something wrong?

tjacobs, 2017-05-25:
The segmentation fault seems to occur in the render() function, and yes, only on the second call.

winfried-loetzsch, 2017-05-31:
I assume this is a bug in the underlying rendering engine, are there plans for fixing it? For me it makes the software unusable..

olegklimov, 2017-05-31:
Won't worry, we'll fix it.

olegklimov, 2017-06-01:
Ha, if you call `env.render("human")` somewhere, it works. Must be some kind of initialization problem. No worries, we'll fix it.

winfried-loetzsch, 2017-06-01:
Thanks for the reply.
Hm, when I call `env.render("human")` in between, I get a different error:

> render-glwidget.cpp:299 ERROR: 0x502
> python3.5: render-glwidget.cpp:299: void Viz::_render_on_correctly_set_up_context(): Zusicherung »e == GL_NO_ERROR« nicht erfüllt.

Only calling `env.render("human")` over multiple episodes also does not work for me like in the agent_zoo example (just for the sake of completeness).

olegklimov, 2017-06-07:
Please test.

closed 2017-06-07 by olegklimov

winfried-loetzsch, 2017-06-10:
Okay, now it does not crash anymore, but the code above produces some weird visualizations after the second call to reset() .. sometimes env.render("rgb_array") returns just background as well.
This is one example:

[image]

olegklimov, 2017-06-11:
Nice to hear it doesn't crash anymore. Are you calling render() just after reset() call, or is there step() in between them?

I've seen similar behavior, mostly in bigger robots. All parts are initialized in center, moved into place after one simulation step. The possible reason is Bullet engine running in separate thread, synchronization problems.

If this problem worth it, please create a new issue.

lrozo, 2017-08-02:
Hi guys,

I am getting a similar error when running any of the example provided in agent_zoo. Whenever the env.step() is called the second time in a loop (after the env.reset()), I get the following error:

assert( np.isfinite(x) )
AssertionError

Which points to the line 88 in gym_pendulums.py. 

I followed the installation instructions provided in the README and everything was OK. Any idea about this issue?

Thanks.

olegklimov, 2017-08-02:
Never crashes for me... If you find a reason for your crashes, please report.

denizs, 2017-09-29:
@olegklimov @lrozo 
Encountered the same issue (#44) on my OSX machine. 
That said, it works for me on ubuntu. Seems to be a problem with osx x roboschool

olegklimov, 2017-09-29:
Hmm can you post the code?

denizs, 2017-09-29:
Ok I ran some more tests. AFAI can tell, the env reset does not work properly due to the render.
First, I ran the code without rendering anything:
```python
for c in range(100000):
    obs, r, d, i = env.step(env.action_space.sample())
    if d:
        env.reset()
        print('{}@{}'.format(d, c+1))
```
Which ran through as expected:
[image]

Next, I pasted in a `render("human")`:
```python
for c in range(100000):
    obs, r, d, i = env.step(env.action_space.sample())
    env.render("human")
    if d:
        env.reset()
        print('{}@{}'.format(d, c+1))
```
The following video shows frames 900-1000. This behaviour is consistent for both, `render("human")` as well as `render("rgb_array")`:

[image]
Lastly, I caught the exception by wrapping `step` into a try except phrase. This lead to the env completely deranging from expected behaviour, and constantly returning `done=True`
```python
for c in range(100000):
    try:
        obs, r, d, i = env.step(env.action_space.sample())
    except:
        pass
    env.render("human")
    if d:
        env.reset()
        print('{}@{}'.format(d, c+1))
```

[image]

## Video Files:
[Without render](https://www.dropbox.com/s/wxdtjn6gzoocbpj/worender.mov?dl=0)
[With human](https://www.dropbox.com/s/exfte0oh75rdhhv/whuman.mov?dl=0)
[With catch](https://www.dropbox.com/s/mz6xzgwg0oxnw6s/wcatch.mov?dl=0)

whoiscris, 2019-06-05:
Any updates on this ? I get obs and act updates as nan values, after I reset the env. But this happens only when using env.render(). I'm using 18.04, and roboschool 1.0.48.

johnmoberg, 2019-07-22:
I have the same issue on 18.04 with Roboschool 1.0.48. @whoiscris Did you figure it out?

PMatthaei, 2019-08-24:
I am also experiencing this problem of nan states after reset with roboschool 1.0.48 on Ubuntu. Only removing `env.render()` helps. Adding `env.render("human")` results in the above mentioned error: QImage::setPixel: coordinate out of range