openai/roboschool #8: OpenGL problem on ubuntu 16.04

asimmunawar, 2017-05-26:
I installed everything according to instructions. However, I am getting the following error. I have tried everything given in https://github.com/openai/roboschool/issues/2. But nothing seems to work. I have been struggling with this problem for some time now. Any help will be greatly appreciated.

```
$ python3 agent_zoo/demo_race2.py 
[2017-05-26 22:20:30,281] Making new env: RoboschoolHumanoid-v0
[2017-05-26 22:20:30,284] Making new env: RoboschoolHumanoid-v0
[2017-05-26 22:20:30,285] Making new env: RoboschoolHumanoid-v0
QGLShaderProgram: could not create shader program
bool QGLShaderPrivate::create(): Could not create shader of type 2.
python3: render-simple.cpp:244: void SimpleRender::Context::initGL(): Assertion `r0' failed.
Aborted (core dumped)

```

I am running

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
```

```
$ nvidia-smi 
Fri May 26 22:27:38 2017       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 375.66                 Driver Version: 375.66                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN X (Pascal)    Off  | 0000:03:00.0     Off |                  N/A |
| 23%   26C    P8    10W / 250W |    192MiB / 12189MiB |      4%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID  Type  Process name                               Usage      |
|=============================================================================|
|    0      1572    G   /usr/lib/xorg/Xorg                              70MiB |
|    0      4455    G   compiz                                         120MiB |
+-----------------------------------------------------------------------------+
```

```
$ glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: NVIDIA Corporation
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    ....
client glx vendor string: NVIDIA Corporation
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    ...
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    ...
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: TITAN X (Pascal)/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 375.66
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_AMD_multi_draw_indirect, GL_AMD_seamless_cubemap_per_texture, 
    GL_AMD_vertex_shader_layer, GL_AMD_vertex_shader_viewport_index, 
    ...
OpenGL version string: 4.5.0 NVIDIA 375.66
OpenGL shading language version string: 4.50 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
    GL_AMD_multi_draw_indirect, GL_AMD_seamless_cubemap_per_texture, 
    GL_AMD_vertex_shader_layer, GL_AMD_vertex_shader_viewport_index, 
    ...
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 375.66
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
    GL_ANDROID_extension_pack_es31a, GL_EXT_base_instance, 
    GL_EXT_blend_func_extended, GL_EXT_blend_minmax, GL_EXT_buffer_storage, 
    GL_EXT_color_buffer_float, GL_EXT_color_buffer_half_float, 
    GL_EXT_copy_image, GL_EXT_deb
```

cjjeakle, 2017-05-27:
I saw this, too. Give these steps a shot: https://github.com/openai/roboschool/issues/6#issuecomment-303972827

asimmunawar, 2017-05-28:
cjjeakle, thanks alot,
Followed the instructions  under "Fix nvidia driver issues". It solved the problem.
So there is some problem with the default nvidia-375 driver.

closed 2017-05-28 by asimmunawar

johnkelly, 2017-07-09:
Seeing this issue (same error) on Ubuntu-16.04 with the latest Nvidia driver 381.22 with 1080TI gpu. Unfortunately, the 381.22 driver is the only driver I can use with my motherboard/gpu.

jacknlliu, 2017-07-31:
This issue still exists for me. I use Ubuntu 16.04 with Nvidia driver 375 with GTX960 GPU.

davidljung, 2017-09-24:
Just adding `from OpenGL import GLU` to the beginning of the pyhon main resolved this issue for me (e.g. to the top of `agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py ` (- this was suggested elsewhere, I don't recall where I saw it).

ethanluoyc, 2017-12-27:
I can confirm that @davidljung's method works for me.

github-kk, 2018-02-01:
I added `from OpenGL import GLU` to the py. It works !!! Ubuntu 16.04  nvidia-381. 
@davidljung  Thanks :)

lutein, 2018-02-28:
@davidljung Thanks a lot! Works on Ubuntu 16.04, GTX 1060, nvidia-390.

chedanix, 2018-04-19:
@davidljung Thank you so much! Works on ubuntu 16.04, GTX 1080, nvidia-384

ouyangzhuzhu, 2018-11-24:
hi~，how  to  add "from OpenGL import GLU" , does it like this below ? ： @davidljung 
from OpenGL import GLU
if __name__=="__main__":
    demo_run()

LUKELIEM, 2019-07-11:
@davidljung's method works for me as well.

I encountered the error when running a code to record an episode of a trained model. I simply add from OpenGL import GLU to the top of the Python code creating the environment. There is no need to add it to the RoboSchool_XXX.py itself.

PierreExeter, 2019-09-04:
@davidljung 's method worked for me, thanks a lot! On Ubuntu 18.04 GTX 1060. I just had to install the pyopengl from Anaconda