openai/roboschool #9: Trouble building

pender, 2017-05-27:
Hi, when I try `pip3 install` on Ubuntu 15.04 with python 3.4, I get the following error:

```
(tf3)pender@penderforge:~/roboschool$ pip3 install -e $ROBOSCHOOL_PATH
Obtaining file:///home/pender/roboschool
    Complete output from command python setup.py egg_info:
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
    sh: 1: install_name_tool: not found
[... 123 lines cut ...]
                ^
    At global scope:
    cc1plus: warning: unrecognized command line option "-Wno-deprecated-register"
    Makefile:128: recipe for target '.build-release/assets-mesh.o' failed
    make: *** [.build-release/assets-mesh.o] Error 1
    make: *** Waiting for unfinished jobs....
    cd /home/pender/roboschool/roboschool/cpp-household && make clean && make -j4 dirs ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pender/roboschool/
```

Can anyone help?

asimmunawar, 2017-05-27:
Did you install the prereqs as written in the README.md?
https://github.com/openai/roboschool/blob/master/README.md

pender, 2017-05-27:
Yes, although apt install couldn't find libpython3.5-dev:

```
pender@penderforge:~/roboschool$ sudo apt install cmake ffmpeg pkg-config qtbase5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libpython3.5-dev
E: Couldn't find any package by regex 'libpython3.5-dev'
```

However I confirmed that libpython3.4-dev is installed and up to date:

```
pender@penderforge:~/roboschool$ sudo apt install libpython3.4-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libpython3.4-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.
```

Santara, 2017-05-28:
I am also facing the exact same issue. Apparently, libpython3.5-dev is not available for Ubuntu 14.04. Can anyone please clarify. All other dependencies are installed.

Zhang-Yong, 2017-05-28:
I have same issue on Ubuntu 14.04, switch to 16.04 the issue will fix.

pender, 2017-05-28:
Thanks @Zhang-Yong , upgraded to 16.04, and then upgraded Nvidia drivers according to @cjjeakle's instructions [here](https://github.com/openai/roboschool/issues/6#issuecomment-303972827) and that worked.

Would be nice if the readme could include a requirements section but I'm not complaining. Thanks again.

closed 2017-05-28 by pender

fairlyme, 2017-07-07:
it's not need to upgrad ubuntu to 16.04, just install assimp from the source.

rdaries29, 2017-07-25:
Hi, I seem to be getting the same issue when I attempt `pip3 install` on my Mac. I get the following error. I have deleted roboschool and re-installed it. The previous time, I had an issue with segmentation errors so I began the process again.

[image]

olegklimov, 2017-07-25:
Now install bullet, as it says in the message.

glazari, 2017-08-01:
I have this same issue on Ubuntu 16.04.  I've installed all the dependencies, but still no luck