openai/roboschool #10: ImportError: libBulletCollision.so.2.87: cannot open shared object file: No such file or directory

jdarpinian, 2017-05-29:
I got this error when trying to run something from the agent zoo after following the installation instructions.

ImportError: libBulletCollision.so.2.87: cannot open shared object file: No such file or directory

To resolve this error I have to do:
export LD_LIBRARY_PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install/lib/
But hopefully there's a way to avoid people having to do that?

olegklimov, 2017-05-29:
Right. You aren't supposed to set it.

```bash
RPATH=-Wl,-z,origin -Wl,-rpath,'$$ORIGIN/cpp-household/bullet_local_install/lib'
```

In Makefile it has path compiled in, starting from `$$ORIGIN` of the binary (second `roboschool` folder). That's why it should find your libraries. If it does not, well, did they broke rpath on Ubuntu 17? Are you installing with `pip3 install -e` or without `-e` (both normally work)? Any trivial typo errors you can find?

olegklimov, 2017-05-30:
@cjjeakle @jdarpinian can you please run:

```bash
LD_DEBUG=libs ldd cpp_household.so
```

On my system I get:

```
     27268:	find library=libBulletDynamics.so.2.87 [0]; searching
     27268:	 search path=/home/user/roboschool/roboschool/./cpp-household/bullet_local_install/lib		(RPATH from file ./cpp_household.so)
     27268:	  trying file=/home/user/roboschool/roboschool/./cpp-household/bullet_local_install/lib/libBulletDynamics.so.2.87
```

Maybe you'll understand what's going on by reading what it searches on your system. 
There is also `-Wl,--enable-new-dtags,-rpath,<dir>` linker instructions I've found [here in this great explanation](http://amir.rachum.com/blog/2016/09/17/shared-libraries/). You can easily try this.

cjjeakle, 2017-05-30:
Thanks @olegklimov! I'm on my phone for now, but will give that a shot tonight.

jdarpinian, 2017-05-30:
Here's the result:
https://gist.github.com/jdarpinian/4520e5821518ffd6bb708d6b8ea40b40

It looks like what's in the .so is runpath, not rpath. Also it looks like the libraries that cpp_household.so depends on directly such as libBulletDynamics.so.2.87 are found, but their further dependencies such as libBulletCollision.so.2.87 are not because there's no runpath in the bullet .so files and they don't inherit the runpath from cpp_household.so apparently.

jdarpinian, 2017-05-30:
Runpath not being used for transitive dependencies is a bug (or at least undocumented behavior): https://bugs.launchpad.net/ubuntu/+source/eglibc/+bug/1253638

I can only guess that the linker in Ubuntu 17.04 has changed the default from --disable-new-dtags to --enable-new-dtags, so we're getting runpath instead of rpath. Explicitly specifying --disable-new-dtags fixes it. I'll send a pull request.

closed 2017-05-31 by olegklimov

jdarpinian, 2017-06-01:
Thanks for the pointer to the rpath stuff, you learn something new every day! It's crazy that they are introducing new bugs in the dynamic linker today. Seems like this stuff should have been all sorted out in the 90s or so...