openai/roboschool #13: Issues with Installation for Anaconda with Python3

mufeili, 2017-05-30:
In the guide of installation for Mac, Anaconda with Python3, 

`brew install cmake tinyxml assimp ffmpeg
brew install boost-python --without-python --with-python3 --build-from-source
conda install qt
export PKG_CONFIG_PATH=$(dirname $(dirname $(which python)))/lib/pkgconfig`

you used brew. I think people rarely use brew and conda together and the underlying python packages are generally different. Would this be a issue? Is there a way to work around brew completely? Thank you.

olegklimov, 2017-05-31:
It works, that's why it's there.

> Is there a way to work around brew completely? 

Possibly, yes. Quick googling produces some references to boost-python in conda, for example. If you can go through installation with only conda packages, let's post your experience for others to follow.

mufeili, 2017-06-01:
@olegklimov Hi Oleg, thanks for the reply. I've tried and I think I can install cmake with conda. However it seems there is no way to install tinyxml without brew for Mac OSX(i.e. with pip/conda/from source, etc.)

olegklimov, 2017-07-18:
This project doesn't need tinyxml. Only Bullet does, and it bundles tinyxml in its repository. You can try again.

jtoy, 2017-08-25:
Im having trouble getting this to work on 10.10.5 . would love to see non brew instructions as well.

cuevasclemente, 2017-09-13:
I had problems with this as well following the instructions on the repo on OSX 10.12.6 using conda.

I was able to resolve the problems by adding my conda lib and include directories to the the environment variable $PKG_CONFIG_PATH. I set my PKG_CONFIG_FLAG as:
`"/usr/local/lib/pkgconfig:/usr/local/lib:$HOME/anaconda/lib/pkgconfig"`.

I also downloaded the version of libboost available on meznom's channel (https://anaconda.org/meznom/boost-python) using the command:
` conda install -c meznom boost-python 

After following the homebrew instructions laid out in the README and doing that, I was able to setup locally no problem with `pip install -e $ROBOSCHOOL_PATH`

After this I was able to get it to run without incident.

stephenjhansen, 2017-10-02:
I am still having some issues and I am not sure if it is a result of brew + Conda. Hopefully I am not polluting the issue thread with too much here.

Any advice would be appreciated.

I am running macOS 10.12.6, I am using a conda environment. I followed the "Mac, Anaconda with Python 3" instructions.

All was well until running pip3:

`Obtaining file:///Users/stephenhansen/Code/DeepRL/gym/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Mac, assumed python3, homebrew branch
    Link against python 3.5
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    /bin/sh: pkg-config: command not found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    /bin/sh: pkg-config: command not found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    /bin/sh: pkg-config: command not found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    /bin/sh: pkg-config: command not found
    In file included from physics-bullet.cpp:1:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/physics-bullet.o] Error 1
    make: *** Waiting for unfinished jobs....
    In file included from assets-mesh.cpp:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/assets-mesh.o] Error 1
    In file included from random-world-tools.cpp:3:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/random-world-tools.o] Error 1
    In file included from render-glwidget.cpp:2:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
Werkzeug                  0.12.2                    <pip>
wheel                     0.30.0                    <pip>
wheel                     0.29.0           py35ha7aa5c4_1  
xz                        5.2.3                ha24016e_1  
zlib                      1.2.11               h60db283_1  `

`(deeprl2) stephenhansen@Stephens-MacBook-Pro:~/Code/DeepRL/gym$ brew list
assimp		cmake		gcc		isl		mpfr		pcre		readline	tinyxml		xvid
boost		ctags		gdbm		lame		openssl		python3		sqlite		wget		xz
boost-python	ffmpeg		gmp		libmpc		openssl@1.1	qt		swig		x264
`

olegklimov, 2017-10-02:
`pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` should run.

stephenjhansen, 2017-10-02:
thank you @olegklimov. After running pkg-config and the following, I was able to run `pip3 install -e $ROBOSCHOOL_PATH` without errors.

although trivial, In case someone else comes across this in a search here are the other things I did get it to install.

> sudo chown -R $USER ~/anaconda/lib/python3.6

I also had to install a few packages I hadn't installed in that environment earlier

> conda install numpy six

I am now having trouble accessing the environments that ship with roboschool.

When I try to execute the suggested test of the flagrun I get a key error:

> python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
> [2017-10-01 23:32:02,421] Making new env: RoboschoolHumanoidFlagrun-v1
> Traceback (most recent call last):
>   File "/Users/stephenhansen/Code/DeepRL/gym/gym/envs/registration.py", line 140, in spec
>     return self.env_specs[id]
> KeyError: 'RoboschoolHumanoidFlagrun-v1'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/Users/stephenhansen/Code/DeepRL/gym/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 492, in <module>
>     demo_run()
>   File "/Users/stephenhansen/Code/DeepRL/gym/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 23, in demo_run
>     env = gym.make("RoboschoolHumanoidFlagrun-v1")
>   File "/Users/stephenhansen/Code/DeepRL/gym/gym/envs/registration.py", line 164, in make
>     return registry.make(id)
>   File "/Users/stephenhansen/Code/DeepRL/gym/gym/envs/registration.py", line 121, in make
>     spec = self.spec(id)
>   File "/Users/stephenhansen/Code/DeepRL/gym/gym/envs/registration.py", line 150, in spec
>     raise error.UnregisteredEnv('No registered env with id: {}'.format(id))
> gym.error.UnregisteredEnv: No registered env with id: RoboschoolHumanoidFlagrun-v1

I am not sure how related this comment is to the intent of this issue (Conda and Brew dependancies) I would be happy to make a new issue if you think it necessary.

stephenjhansen, 2017-10-02:
To clarify the problem I am still having as mentioned in the previous post.

- roboschool installs without errors
- I can run import roboschool without errors
- I cannot run any of the environments that ship with roboschool 
  e.g. `env2 = gym.make('RoboschoolHumanoidFlagrun-v1')` returns
  gym.error.UnregisteredEnv: No registered env with id: RoboschoolHumanoidFlagrun-v1

Here are the install steps I have completed so far:

1. Mac, Anaconda with Python 3 instructions from README
2. Compile and install bullet instructions from README
3.` sudo chown -R $USER ~/anaconda/lib/python3.6` to prevent ownership issues
4. `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5 `as suggested by @olegklimov 

I am running on macOS 10.12.6

Any advice is appreciated. Let me know if I should create a new issue - sorry if I have polluted this thread with unrelated comments.

olegklimov, 2017-10-02:
Oh even if it is like you describe, should be easy to debug. Just with `print`s is alright. Put one in roboschool __init__, put another in Gym.

closed 2017-10-02 by olegklimov

bnjasim, 2017-11-28:
I got the a similar error while `pip install -e $ROBOSCHOOL_PATH`

`pip install -e $ROBOSCHOOL_PATH
Obtaining file:///Users/bnjasim/Desktop/RL/roboschool
    Complete output from command python setup.py egg_info:
    /Users/bnjasim/Desktop/RL/roboschool/roboschool/cpp-household/bullet_local_install/lib
    ******************************
    
    ['libBullet2FileLoader', 'libBulletCollision', 'libBullet3Collision', 'libBulletDynamics', 'libBullet3Common', 'libBulletInverseDynamics', 'libBullet3Dynamics', 'libBulletSoftBody', 'libBullet3Geometry', 'libLinearMath', 'libBullet3OpenCL_clew', 'libPhysicsClientC_API']
    ******************************
    
    found resource files: 185
    running egg_info
    Mac, assumed python3, homebrew branch
    Link against python 3.6
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Mac, assumed python3, homebrew branch
    Link against python 3.6
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-hud.cpp  -o.build-release/render-hud.o -MMD -MF .build-release/render-hud.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_loca [... cut ...]
    make: *** Waiting for unfinished jobs....
    cd /Users/bnjasim/Desktop/RL/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    `

It looks like almost impossible for me to get the roboschool running on my mac. Any help would be appreciated.

YunqiuXu, 2018-01-16:
@stephenjhansen have you solved it? Same problem when installing it with ubuntu 16.04 and anaconda 3