openai/roboschool #18: Windows support for roboschool

ViktorM, 2017-06-02:
Can you add a Windows support for roboschool too?

olegklimov, 2017-06-07:
https://github.com/openai/roboschool/wiki/Help-Wanted

ViktorM, 2017-06-11:
I see. Currently I'm looking into Windows support for rllab. After finishing can take a look into Windows support of Roboschool.

TheCrazyT, 2017-12-23:
Oh well it really seems to be a pain to develop windows support.
I was able to successfully create a dll for cpp_household under msys and the setup.py installs it.
The "only" problem i have now is to get tensorflow working under mingw64+msys2.
At the moment i'm kinda stuck because of that.

Btw.: 
Does somebody know a free windows build server?
I tried it with travis-ci, but msys2 won't properly work with wine.
Also tried tea-ci, but that fails ,too.

TheCrazyT, 2017-12-26:
Ok, after getting through all the pain i managed to create a working compiled version.
You can download it here:

https://www.dropbox.com/s/6tiwitcmahssvew/WinRoboschool.7z?dl=0

Had all kinds of trouble, from compiling the boost-library to some possible mingw-bug.
But atleast we now have a proof of concept that it is possible to make it work in windows.
You probably can still expect some bugs (for example i'm getting a crash message after exiting python).
But the bot-race-demo itself seems to be working flawlessly (beside some graphic glitches).

I compiled it with MinGW + Codeblocks + Msys (currently ignoring the Makefile and setup.py).
Might modify my forked repo later so that its possible to build with the common process.
Here is a small list of the versions:
boost: 1.66.0
mingw-w64-x86_64-assimp: 4.0.1-1
mingw-w64-x86_64-qt5: 5.10.0-1
mingw: x86_64-7.2.0-posix-seh-rt_v5-rev1
msys: msys2-x86_64-20161025.exe

sighingnow, 2017-12-27:
直接编译sample-master（命令：`make master`）也有同样的问题吗？

ZhiqingXiao, 2019-03-13:
@TheCrazyT Thank you for your work on making this wonderful codes install-able on Windows. Could you please explain how to install the latest release of TheCrazyT/roboschool in Windows? Thanks!

TheCrazyT, 2019-03-13:
I think those dependencies are not really  missing because they are windows internal.There was an explanation on stackoverflow i think. (linked that here:https://github.com/TheCrazyT/roboschool/issues/7)

Did you try to modify rundemo.bat and run that ?

ZhiqingXiao, 2019-03-14:
@TheCrazyT Thank you for your prompt reply. Finally the problem is resolved and the demo works. Thanks again for providing Windows support.

(Detail: I later realized that when I used Python 3.7, I also missed `PYTHON36.DLL` and `EMCLIENT.DLL`. So I switched to python 3.6. Then there is an error saying that `c:/tmp` is missing. Then I created the empty folder `c:/tmp` manually. Then the demo works!)
(The gym has been updated in recent months, for example, the design of gym.Env from_step() and _reset() to step() and reset(), which triggers lots of warnings.)

TheCrazyT, 2019-03-16:
Thanks for reminding me, I guess I never thought about the "c:/tmp"-directory.
It's clearly a bug currently that needs to be fixed. 
Normaly it should use the already existing windows temp-folder.
Maybe should also think about adding different python versions ...

SebastianNagy, 2022-06-10:
I'm currently stuck at the "missing DLLs" step. I copied all the dlls into C:\Windows\system32 but I still get the same error. Could someone please tell me how to fix this? I've been trying to get it work for a whole day now....

TheCrazyT, 2022-06-10:
@SebastianNagy 
I'm sorry to say that, but I guess you can't even get up2date dll's at the moment.
(since bintray seem to have changed alot over the years and my login somehow got lost because of that).

I probably need to change the deployment, but I must admit that I didn't modify my windows fork for years.
And I'm not going to, since this project itself even claims that it is "DEPRECATED".

SebastianNagy, 2022-06-10:
> @SebastianNagy I'm sorry to say that, but I guess you can't even get up2date dll's at the moment. (since bintray seem to have changed alot over the years and my login somehow got lost because of [...]

I know that it says deprecated but I need to use SLM Lab for a university project. I managed to get it to work on another laptop where I have Ubuntu, but no GPU. And I'd also like it to work on Windows, where I have a GPU

SebastianNagy, 2022-06-10:
But maybe it's just something I did wrong. Could you tell me the steps I should have done 3 years ago?

I downloaded your archive, unarchived, copied roboschool in the python env packages, copied the dlls in windows/system32. Anything else I missed?

TheCrazyT, 2022-06-10:
@SebastianNagy 

Are you using the version mentioned in the dropbox-link?
No need to pollute your system32-folder, the version in the dropbox-link should work just fine by modifying the paths inside the rundemo.bat .
Only dependency should be the right python-version (3.6) if i remember correctly.
You can't use anything newer because the DLL's were compiled against it.

SebastianNagy, 2022-06-10:
Oh I see. I am using Python 3.7. Will give it a go with python 3.6. Thank you!

SebastianNagy, 2022-06-10:
A python3.6 env got me past the roboschool error, however, now I get this: Sanity check failed: your Qt4 installation is broken. You can try to fix it by export QT_PLUGIN_PATH=/usr/local/Cellar/qt/4.8.7_2/plugins

PyQt4 is installed, not sure what's wrong..

TheCrazyT, 2022-06-11:
@SebastianNagy 
It is probably missing some DLL's as dependencies.
I found a more up to date version and uploaded it to dropbox: https://www.dropbox.com/s/9f2zt9ql6u8u88g/WinRoboschool3.zip?dl=0
But keep in mind that you need to update all the paths in the *.bat files.
(that involves paths of installed python, cuda and cuDNN)

SebastianNagy, 2022-06-11:
> @SebastianNagy It is probably missing some DLL's as dependencies. I found a more up to date version and uploaded it to dropbox: https://www.dropbox.com/s/9f2zt9ql6u8u88g/WinRoboschool3.zip?dl=0 But [...]

You kind sir, are a God amongst men! It works on window, thank you so so so so so much! I really wish I could somehow repay you!

TheCrazyT, 2022-06-11:
@SebastianNagy 
I'm glad it helps you atleast a little bit.
But some parts can still be buggy.

I think the last thing i tried to fix were the examples where agents play against each other.
I'm not really shure if everything is fixed on that part.
(rundemoMP.bat and rundemoMP3.bat probably won't work ...)