openai/roboschool #19: macOS not finding boost

peabody124, 2017-06-04:
I've installed all the prereqs under `Mac, homebrew python`. However when I try and install (or compile cpp-household) the include directories aren't pulling in /usr/local/include (which would grab boost)

```
:~/python/roboschool/roboschool/cpp-household (master)$ make
Mac, assumed python3, homebrew branch
Link against python 3.6
Slow hardware or software render (no shadows)
gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
In file included from physics-bullet.cpp:1:
In file included from ./render-glwidget.h:1:
In file included from ./render-simple.h:3:
In file included from ./household.h:2:
./assets.h:7:10: fatal error: 'boost/shared_ptr.hpp' file not found
#include <boost/shared_ptr.hpp>
         ^
1 error generated.
```

and it doesn't come in from the pkg-config line:

```
$ pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6
-I/usr/local/Cellar/qt/5.9.0/include/QtOpenGL -I/usr/local/Cellar/qt/5.9.0/include -I/usr/local/Cellar/qt/5.9.0/include/QtWidgets -I/usr/local/Cellar/qt/5.9.0/include -I/usr/local/Cellar/qt/5.9.0/include/QtGui -I/usr/local/Cellar/qt/5.9.0/include -I/usr/local/Cellar/qt/5.9.0/include/QtCore -I/usr/local/Cellar/qt/5.9.0/include -I/usr/local/Cellar/assimp/3.3.1/include/assimp -I/usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/include/python3.6m
```

boost is installed

```
$ brew list | grep 'boost'
boost
boost-python
boost@1.60
$ ls /usr/local/include/boost/shared_ptr.hpp 
/usr/local/include/boost/shared_ptr.hpp
```

I can fix the compilation with this diff on the Makefile, but wasn't sure what I'm missing this is working for others

```
diff --git a/roboschool/cpp-household/Makefile b/roboschool/cpp-household/Makefile
index 3b5c879..a32f847 100644
--- a/roboschool/cpp-household/Makefile
+++ b/roboschool/cpp-household/Makefile
@@ -38,7 +38,7 @@ ifeq ($(UNAME),Darwin)
   endif
   PKG   =pkg-config
   LIBS  =-framework OpenGL
-  INC   =-I/System/Library/Frameworks/OpenGL.framework/Headers
+  INC   =-I/System/Library/Frameworks/OpenGL.framework/Headers -I/usr/local/include
   BOOST_PYTHON3_POSTFIX = 3
   BOOST_MT=-mt
   PYTHON = $(shell python3 -c "import sys; print('%i.%i' % sys.version_info[:2])")
```

jruales, 2017-06-04:
I had the same issue. After applying your fix, it's able to find boost, but then I get `fatal error: 'QtWidgets/QOpenGLWidget' file not found`.

peabody124, 2017-06-04:
I had to make a few more changes to get it fully functional (again this is macOS 10.12.15). It seems like the location of openGL has changed, and also the path to tinyxml was not found.

```
diff --git a/roboschool/cpp-household/Makefile b/roboschool/cpp-household/Makefile
index 3b5c879..54b8b36 100644
--- a/roboschool/cpp-household/Makefile
+++ b/roboschool/cpp-household/Makefile
@@ -38,7 +38,7 @@ ifeq ($(UNAME),Darwin)
   endif
   PKG   =pkg-config
   LIBS  =-framework OpenGL
-  INC   =-I/System/Library/Frameworks/OpenGL.framework/Headers
+  INC   =-I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/usr/local/include
   BOOST_PYTHON3_POSTFIX = 3
   BOOST_MT=-mt
   PYTHON = $(shell python3 -c "import sys; print('%i.%i' % sys.version_info[:2])")
@@ -47,7 +47,7 @@ endif
 
 $(info Link against python $(PYTHON))
 INC     += `$(PKG) --cflags Qt5Widgets Qt5OpenGL assimp python-$(PYTHON)`
-LIBS    += -lstdc++ -ltinyxml `$(PKG) --libs Qt5OpenGL Qt5Widgets assimp python-$(PYTHON)`
+LIBS    += -lstdc++ `$(PKG) --libs Qt5OpenGL Qt5Widgets tinyxml assimp python-$(PYTHON)`
 INC     += -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet
 LIBS    += $(RPATH) -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API
```

olegklimov, 2017-06-11:
Hi @peabody124, very nice effort compiling on newer macOS versions! I think homebrew will eventually adapt to some of this changes, especially .pc files.

And this path `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Headers` looks crazy to me. There must be a better way.

Please post here any new things you encounter, it's valuable. Developer beta will turn normal version some day.

rdaries29, 2017-07-07:
I still seem to be getting the file reference error for QtWidgets/QOpenGLWidget' file not found as@ jruales had even after updating the MakeFile paths.
[image]

erwincoumans, 2017-08-16:
I can confirm I had to patch all the various parts in the Makefile, related to OpenGL, boost and qt. After a few more fiddling, it finally compiled and ran on Mac OSX Sierra 10.12.6 (regular release, not Beta).

nemanja-rakicevic, 2017-08-18:
Hi @erwincoumans could you please copy your Makefile here, I'm still having problems as lboost_python cannot be found...

erwincoumans, 2017-08-18:
Sure, see attached Makefile. This is using the brew install python3 (not the Anaconda version).

brew info boost-python

erwincoumans-macbookpro2:cpp-household erwincoumans$ brew info boost-python
boost-python: stable 1.64.0 (bottled), HEAD
C++ library for C++/Python interoperability
https://www.boost.org/
/usr/local/Cellar/boost-python/1.64.0 (458 files, 16.9MB) *
  Built from source on 2017-08-16 at 12:18:00 with: --without-python --with-python3
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost-python.rb
==> Dependencies
Required: boost ✔
==> Requirements
Optional: python3 ✔
==> Options
--c++11
	Build using C++11 mode
--with-python3
	Build with python3 support
--without-python
	Build without python 2 support
--HEAD

Also I used:
874  [2017-08-16 14:11:14 -0700] export PATH=/usr/local/bin:/usr/local/opt/qt5/bin:$PATH
875  [2017-08-16 14:11:20 -0700] export PKG_CONFIG_PATH=/usr/local/opt/qt5/lib/pkgconfig

[Makefile.zip](https://github.com/openai/roboschool/files/1235728/Makefile.zip)

nemanja-rakicevic, 2017-08-18:
hmm, i still get this error:
`ld: library not found for -lboost_python3
    clang: error: linker command failed with exit code 1 (use -v to see invocation)`

It's like the name is wrong, I tried `mdfind boost_python3` but it doesn't seem to find it. Am I doing something wrong?

nemanja-rakicevic, 2017-08-23:
Solved it by reinstalling boost-python.

trickmeyer, 2017-08-24:
Boost 1.65 was released two days ago, and all this is dependent upon 1.64.  The repo's homebrew instructions will install this new version, and even with the above it won't work.  Trying to get it working now - I'll update if I have any luck.

nemanja-rakicevic, 2017-08-25:
installation went through, but cannot test:
`$ python $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py`
gives an error:
`Traceback (most recent call last):
  File "/Users/nemanja/Workspace/src/roboschool//agent_zoo/demo_race2.py", line 1, in <module>
    import os, gym, roboschool
  File "/Users/nemanja/Workspace/src/roboschool/roboschool/__init__.py", line 92, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/Users/nemanja/Workspace/src/roboschool/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/Users/nemanja/Workspace/src/roboschool/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: dlopen(/Users/nemanja/Workspace/src/roboschool/roboschool/cpp_household.so, 2): Library not loaded: @rpath/libQt5OpenGL.5.dylib
  Referenced from: /Users/nemanja/Workspace/src/roboschool/roboschool/cpp_household.so
  Reason: image not found`

urbanintell, 2017-09-15:
What folder does the MakeFile go in?

stephenjhansen, 2017-10-02:
sorry to jump in here - but I want to clarify this issue. Is macOS 10.12.6 not supported for roboschool v1.1? I am not able to get roboschool installed and have spent a bit of time trying to figure out a work around.

erwincoumans, 2017-10-03:
Not sure what you want to achieve, but if you want a really simple installation, use the following command: pip install pybullet. Then visit http://pybullet.org and see how to use it, there is a small section about Reinforcement Learning, and some of the Roboschool are already converted.

vglazer, 2017-10-29:
I'm also on macOS 10.12.6, running Python 3.6. I was able to build roboschool without error by following the "Mac, Anaconda with Python 3" section of the installation instructions, but python dumps core with `Segmentation fault: 11` as soon as I `import roboschool`. I haven't had a chance to dig deeper into what's causing this, but if I figure it out I will report back with my findings.

I did manage to `pip install pybullet` without incident, just like @erwincoumans said, and am able to run the Roboschool environments it ships with.