openai/roboschool #23: Boost Python Error

christiansafka, 2017-06-08:
Hi, I am getting

`Boost.Python.ArgumentError: Python argument types in
    World.set_glsl_path(World, str)
did not match C++ signature:
    set_glsl_path(World {lvalue}, std::string)
`

when attempting to run the demo.  Here is the relevant output from ' dpkg --list | grep libboost ':

` amd64        Boost.Python Library development files (default version)
ii  libboost-python1.58-dev                      1.58.0+dfsg-5ubuntu3.1                                amd64        Boost.Python Library development files
ii  libboost-python1.58.0                        1.58.0+dfsg-5ubuntu3.1                                amd64        Boost.Python Library
ii  libboost-random-dev:amd64                    1.58.0.1ubuntu1     `

Is it because my libboost is likely compiled against python2.7 ?

olegklimov, 2017-06-11:
No. Here is files list:

https://packages.ubuntu.com/xenial/amd64/libboost-python1.58-dev/filelist

it provides both 2.7 and 3.5 versions.

Please provide more instructions on how to reproduce your situation, so I can help you (and others) more.

songrotek, 2017-06-12:
I have the same problem! i use python2.7 to install roboschool. After installed successfully, i ran the agent_zoo examples and got the same problem above.

olegklimov, 2017-06-12:
Great! Let's fix it. @songrotek @christiansafka can you describe how to reproduce the problem?

songrotek, 2017-06-12:
I just follow the README to install the roboschool. It seems that everything is ok. I tried python 2.7 and python 3.5, both failed in the same problem.

olegklimov, 2017-06-12:
So what's the starting point? OS? Conda? Linuxbrew?

christiansafka, 2017-06-12:
I'm on Ubuntu 16.04
Python 3.5.2
GCC 4.9.4 ( but if I run python3 I can see GCC 5.4.0 -- not sure if it's relevant )

songrotek, 2017-06-13:
im on Ubuntu 16.04 and python 2.7

immars, 2017-06-16:
same problem here, also ubuntu 16.04 and python 2.7, in virtualenv

also tried python 3.5 but no luck.

Pistoletek, 2017-07-05:
I'm on ubuntu 16.04, python 3.5, boost 1.58. Updating boost to 1.64 solved this issue for me.
Now I am stuck at OpenGL error #27, but I don't think it is related to this issue in any way.

immars, 2017-07-12:
@marek-danel thanks! ubuntu 16.04, python 2.7 in virtualenv, boost 1.64 worked for me too.

gl related problem I encountered is 

```
QGLShaderProgram: could not create shader program
bool QGLShaderPrivate::create(): Could not create shader of type 2.
```

solved by upgrading nvidia driver to 381.09 from 375.66.

mphielipp, 2017-10-04:
How you guys install boost 1.64? I tried the instructions but I get still stuck with the version 1.58. (I try purge , etc, and also follow the install step in boost)
I do have conda, and Ubuntu 16.04.

paive, 2017-10-30:
@mphielipp First download boost from [http://www.boost.org/users/download/](url), you just need run two commands in the directory of boost:  

- `./bootstrap.sh`

- `./b2  install`.

It will install automatically in '/usr/local/include', so you need to be the administrator.

Next, edit you .bashrc file adding 

- `export  BOOST_INCLUDE=/usr/local/include`

- `export  BOOST_LIB=/usr/local/lib`

Find a example of boost python and test if it works, if it occurs the error such that
`In file included from /usr/local/include/boost/python/detail/prefix.hpp:13:0,
                 from /usr/local/include/boost/python/args.hpp:8,
                 from /usr/local/include/boost/python.hpp:11,
                 from hello_ext.cpp:2:
/usr/local/include/boost/python/detail/wrap_python.hpp:50:23: fatal error: pyconfig.h:`
just add`export CPLUS_INCLUDE_PATH=/usr/include/python2.7` to .bashrc.

It works to me,  I hope this is useful for you

DanteLuo, 2017-11-03:
@paive Thanks for your instruction. I did the same thing but the error is still there. I am using python3.5 ubuntu and pip. 
I checked the dir of 1.64 boost and add to the export path at the end of bashrc. But it doesn't work.

Is there any way to around this?

marianophielipp, 2017-11-10:
does not work. I'm trying to use a docker image.