openai/roboschool #24: Module fails to load in Python 2.7

jmessias, 2017-06-08:
I have both Python2 and Python3 in my system, as I need the latter to allow certain applications to run, but I don't want to have to use Python3 with roboschool.

I used pip and not pip3 to install roboschool, but found that the module fails to load in python 2.7, as shown here:

[image]

I believe that this is because the Makefile in cpp-household/ forces the usage of python3 as long as it is in the system, and consequently the libs are always linked against libboost_python3 in that case.

A fix which worked for me is here: https://github.com/jmessias/roboschool/commit/dcbb0d176d5411d6a9b02885e67d0ff3762f6198, which allows the user to force python2 usage by setting USE_PYTHON2=1 prior to calling pip install (e.g. by calling USE_PYTHON2=1 pip install -e $ROBOSCHOOL_PATH), but still uses python3 by default.

olegklimov, 2017-06-11:
Hm. What would be the right way to do it... :thinking: 

Maybe pass PY3=0 to make from setup.py?

@jmessias Do you care to make a proper fix? By "proper" I mean, if you run `pip install` it runs setup.py in python2, that in turn runs make with arguments, but still allows to run make manually.

Ah well, let me do it...

closed 2017-06-11 by olegklimov

olegklimov, 2017-06-11:
@jmessias can you please test my fix for this. What I can do on my system is:

```bash
$ pip3 install -e .
$ python3 agent_zoo/RoboschoolHopper_v0_2017may.py
$ pip2 install -e .
$ python2 agent_zoo/RoboschoolHopper_v0_2017may.py
```

both work, but not simultaneously (for obvious reasons).

Outside of scope of this issue, I can also use install into site-packages (without -e) on my linux box. Please also test that if you care.

jmessias, 2017-06-12:
@olegklimov your fix works (with -e), thanks!

However, curiously enough, it doesn't work on my box without editable mode. An install into dist-packages (Debian's default instead of site-packages) with pip2 gives me the same problem as  before.

olegklimov, 2017-06-12:
Great! Now try to copy one of zoo scripts outside of roboschool source tree, so there is no roboschool folder in current dir, parent dir, parent of parent dir or anywhere up. And try to run it again. It should work, because it 2.7 imports that directory instead of installed module. No such behavior in Python3.

commit ae0f43aa (2017-06-11): Support for Python 2.7, fixes #24