openai/roboschool #25: Issue when installing on Ubuntu 16.04

artsuzy, 2017-06-09:
When I try to install Roboschool I get this error: Directory '/path/to/roboschool' is not installable. File 'setup.py' not found. I follow all instructions, my OS is Ubuntu 16.04, can someone help me?

closed 2017-06-09 by artsuzy

lrozo, 2017-06-26:
Hi Artsuzy,

I got the same error, how did you solve it?

wyqsnddd, 2017-06-29:
I solved it by setting '/path/to/roboschool' to where you git cloned the "bullet3" repo. Basically that's a pip-installable python package. 

I think the error is due to the fact that I pointed  '/path/to/roboschool' to the library generated by the "bullet3" repo instead of the "bullet3" repo itself.

lrozo, 2017-06-29:
Yeap, I made a similar mistake... Somehow I messed up things with the two repos that I git-cloned (roboschool and bullet3). But I already solved it. 

Thank you for answering!

nikemr, 2017-07-11:
Hello sorry to bother you but I am a beginner. Could you please explain how do you set the '/path/to/roboschool' to bullet3 repo?
here is my home folder
Thank you
[image]

lrozo, 2017-07-11:
Hi Nikemr, 

When you git clone the repo https://github.com/openai/roboschool in your folder (let's say you git clone it in Documents), then your "/path/to/roboschool" would be ~/Documents/roboschool. 

However, my suggestion is to create a folder in your computer for your code projects (or git repositories). You can create a "Git" folder wherever you want, and then inside you can git clone https://github.com/openai/roboschool 

Once you have git-cloned roboschool, follow the instructions as suggested in the repo (where you are asked to git-clone the bullet3 project).

Best,

Leonel

nikemr, 2017-07-11:
Thank you Irozo will try tomorrow.

sivaram10, 2017-11-25:
Hi Irozo,

I'm a beginner to RL. I'm trying to install roboschool as you mentioned but still getting some errors. I'm not sure where I did mistakes.

1. created a new directory Git 
2. git-cloned roboschool
4. ROBOSCHOOL_PATH=/path/to/roboschool
5. git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision
mkdir bullet3/build
cd    bullet3/build
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..
make -j4
sudo make install
cd ../..

6. pip3 install -e $ROBOSCHOOL_PATH

Getting following error
("Please follow instructions in README to build local (not global for your system) Bullet installation.")

-- This error message is in setup.py in roboschool.

TIA