openai/roboschool #26: some packages not found when install

hoyeunglee, 2017-06-09:
sudo apt-get install cmake ffmpeg pkg-config qtbase5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package ffmpeg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ffmpeg' has no installation candidate
E: Unable to locate package libpython3.5-dev
E: Couldn't find any package by regex 'libpython3.5-dev'
mas@mas-STCK1A8LFC:~/Downloads/reaver-1.4/src$ 
mas@mas-STCK1A8LFC:~/Downloads/reaver-1.4/src$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.2 LTS
Release:	14.04
Codename:	trusty

olegklimov, 2017-06-11:
```
ffmpeg was replaced in previous releases by the avconv program from the libav project.

To install avconv you need to install the libav-tools package:

sudo apt-get install libav-tools
avconv is very similar to ffmpeg, but they also have some differences in syntax.

Edit: It seems Ubuntu 16.04LTS has again ffmpeg in their repositories, 
so libav-tools is a dummy package and only creates an alias
avconv -> ffmpeg (and others)
```

That's what I've found at:

https://askubuntu.com/questions/432542/is-ffmpeg-missing-from-the-official-repositories-in-14-04

@hoyeunglee please try some workarounds, if it works for you, please post here a note what exactly you did to make it work, so we can link this issue from README.

hoyeunglee, 2017-06-13:
mas@mas-STCK1A8LFC:~$ sudo apt install cmake pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libpython3.5-dev
E: Couldn't find any package by regex 'libpython3.5-dev'

got another package missing

olegklimov, 2017-06-13:
Well it's older Ubuntu, it has Python 3.4.

danijar, 2017-07-19:
On Ubuntu 14, replace `libpython3.5-dev` by `libpython3-dev` and `ffmpeg` by `libav-tools`.

andrewliao11, 2017-07-21:
hi, is there anybody success in Ubuntu14.04?

hoyeunglee, 2017-07-24:
rms/Windows.o build/temp.linux-x86_64-3.4/examples/ThirdPartyLibs/Gwen/Renderers/OpenGL_DebugFont.o build/temp.linux-x86_64-3.4/examples/ThirdPartyLibs/enet/unix.o build/temp.linux-x86_64-3.4/examples/OpenGLWindow/X11OpenGLWindow.o build/temp.linux-x86_64-3.4/examples/ThirdPartyLibs/Glew/glew.o -ldl -lpthread -o build/lib.linux-x86_64-3.4/pybullet.cpython-34m.so
    Creating /usr/local/lib/python3.4/dist-packages/pybullet.egg-link (link to .)
    Adding pybullet 1.0.2 to easy-install.pth file
    
    Installed /home/mas/Documents/roboschool
Successfully installed pybullet
Cleaning up...
mas@mas-STCK1A8LFC:~/Documents$ python $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py
python: can't open file '/home/mas/Documents/roboschool/agent_zoo/demo_race2.py': [Errno 2] No such file or directory
mas@mas-STCK1A8LFC:~/Documents$ echo $ROBOSCHOOL_PATH
/home/mas/Documents/roboschool
mas@mas-STCK1A8LFC:~/Documents$ python $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py
python: can't open file '/home/mas/Documents/roboschool/agent_zoo/demo_race2.py': [Errno 2] No such file or directory
mas@mas-STCK1A8LFC:~/Documents$ python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
python: can't open file '/home/mas/Documents/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py': [Errno 2] No such file or directory

i succeed to install but no this library, and how to use?

mas@mas-STCK1A8LFC:~/Documents/roboschool/docs$ python
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import roboschool, gym; print("\n".join(['- ' + spec.id for spec in gym.envs.registry.all() if spec.id.startswith('Roboschool')]))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named roboschool

zacwellmer, 2017-09-03:
Here is a [link](https://www.faqforge.com/linux/how-to-install-ffmpeg-on-ubuntu-14-04/) to getting ffmpeg on ubuntu14

in summary:
```bash
sudo add-apt-repository ppa:mc3man/trusty-media
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install ffmpeg
```

FulltronService, 2019-04-05:
> Here is a [link](https://www.faqforge.com/linux/how-to-install-ffmpeg-on-ubuntu-14-04/) to getting ffmpeg on ubuntu14
> 
> in summary:
> 
> ```shell
> sudo add-apt-repository ppa:mc3man/trusty-media
> sudo apt-get update
> sudo apt-get dist-upgrade
> sudo apt-get install ffmpeg
> ```

Perfect bro, I love you xd