openai/roboschool #27: OpenGL error

artsuzy, 2017-06-09:
When I try to run demo I get this error
```
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform offscreen surface, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
```

rmst, 2017-06-09:
Same for me. Tried to add `from OpenGL import GL` to the script but didn't help.

artsuzy, 2017-06-10:
@olegklimov Can you help?

olegklimov, 2017-06-11:
Hello @artsuzy, any info on how to reproduce this? 

Does tuxracer work? If tuxracer works but this project does not, it means we have a problem.

If tuxracer doesn't, it just means something is wrong with your setup, not a bug in this project.

rmst, 2017-06-11:
For me at least extremetuxracer works (tuxracer wasn't in the Ubuntu source list).

`glxinfo` outputs:
```
name of display: :1
display: :1  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2)  (0x1616)
    Version: 11.2.0
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2) 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 11.2.0
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_3DFX_texture_compression_FXT1, GL_AMD_conservative_depth, 
    GL_AMD_draw_buffers_blend, GL_AMD_seamless_cubemap_per_texture, 
[... 171 lines cut ...]
OpenGL ES profile extensions:
    GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
    GL_APPLE_texture_max_level, GL_EXT_blend_func_extended, 
    GL_EXT_blend_minmax, GL_EXT_buffer_storage, GL_EXT_color_buffer_float, 
    GL_EXT_discard_framebuffer, GL_EXT_draw_buffers, 
    GL_EXT_draw_elements_base_vertex, GL_EXT_map_buffer_range, 
    GL_EXT_multi_draw_arrays, GL_EXT_read_format_bgra, 
    GL_EXT_separate_shader_objects, GL_EXT_shader_integer_mix, 
    GL_EXT_shader_samples_identical, GL_EXT_texture_compression_dxt1, 
    GL_EXT_texture_filter_anisotropic, GL_EXT_texture_format_BGRA8888, 
    GL_EXT_texture_rg, GL_EXT_texture_type_2_10_10_10_REV, 
    GL_EXT_unpack_subimage, GL_KHR_context_flush_control, GL_KHR_debug, 
    GL_NV_draw_buffers, GL_NV_fbo_color_attachments, GL_NV_read_buffer, 
    GL_NV_read_depth, GL_NV_read_depth_stencil, GL_NV_read_stencil, 
    GL_OES_EGL_image, GL_OES_EGL_image_external, GL_OES_EGL_sync, 
    GL_OES_compressed_ETC1_RGB8_texture, GL_OES_depth24, GL_OES_depth_texture, 
    GL_OES_depth_texture_cube_map, GL_OES_draw_elements_base_vertex, 
    GL_OES_element_index_uint, GL_OES_fbo_render_mipmap, 
    GL_OES_get_program_binary, GL_OES_mapbuffer, GL_OES_packed_depth_stencil, 
    GL_OES_rgb8_rgba8, GL_OES_standard_derivatives, GL_OES_stencil8, 
    GL_OES_surfaceless_context, GL_OES_texture_3D, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_texture_storage_multisample_2d_array, GL_OES_vertex_array_object
```

olegklimov, 2017-06-12:
So what exactly did you do wrong? Can we put this in README as something to avoid?

artsuzy, 2017-06-12:
I have used conda environment, is seems that roboschool doesn't work with conda

olegklimov, 2017-06-12:
It works on Linux with Anaconda.

Here is Dockerfile I use:

```
FROM nvidia/cuda:8.0-cudnn5-devel-ubuntu16.04
RUN echo "XKBMODEL=\"pc104\"" >> /etc/default/keyboard
RUN echo "XKBLAYOUT=\"us\"" >> /etc/default/keyboard
ENV LANG=C.UTF-8 LC_ALL=C.UTF-8 DEBIAN_FRONTEND=noninteractive

RUN apt-get update -q && apt-get install -y \
        cmake build-essential autoconf curl libtool libboost-python-dev unzip rsync gdb \
        wget bzip2 ca-certificates \
        libglib2.0-0 libxext6 libsm6 libxrender1 git \
        qtbase5-dev libboost-dev libassimp-dev pkg-config libtinyxml-dev \
        xserver-xorg-video-dummy net-tools x11-apps git wget grep sed dpkg \
        openssh-server ssh-askpass \
        zlib1g-dev libglu1-mesa-dev libglu1-mesa libgl1-mesa-dev xvfb libxinerama1 libxcursor1 libxrandr2 libxi6 \
        libav-tools && rm -rf /var/lib/apt/lists/*

RUN echo 'export PATH=/opt/conda/bin:$PATH' > /etc/profile.d/conda.sh && \
    wget --quiet https://repo.continuum.io/miniconda/Miniconda3-4.1.11-Linux-x86_64.sh -O ~/miniconda.sh && \
    /bin/bash ~/miniconda.sh -b -p /opt/conda && \
    rm ~/miniconda.sh
RUN TINI_VERSION=`curl https://github.com/krallin/tini/releases/latest | grep -o "/v.*\"" | sed 's:^..\(.*\).$:\1:'` && \
    curl -L "https://github.com/krallin/tini/releases/download/v${TINI_VERSION}/tini_${TINI_VERSION}.deb" > tini.deb && \
    dpkg -i tini.deb
ENTRYPOINT [ "/usr/bin/tini", "--" ]
CMD [ "/bin/bash" ]
ENV HOME=/root \
    TINKERBELL_CODE_DIR=/root/code \
    PYTHONPATH=/root/code/tinkerbell:/root/code/rl-algs:/root/code/gym:/root/code/openai-benchmark:/root/code/openai-tinkerbell:/root/code/roboschool \
    PATH=/opt/conda/bin:$PATH \
    LD_LIBRARY_PATH=/usr/local/cuda/lib64:/usr/local/cuda/extras/CUPTI/lib64:/usr/local/cuda/targets/x86_64-linux/lib:/usr/lib/x86_64-linux-gnu:/usr/local/lib
RUN conda install -y numpy scipy opencv && conda install -y -c mpi4py mpich mpi4py

ENV PKG_CONFIG_PATH=/opt/conda/lib/pkgconfig:/usr/lib/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig
RUN pip install \
    tqdm \
    cloudpickle \
    boto3 \
    awscli \
    keras \
    tensorflow \
    mujoco_py atari_py \
    pyglet \
    joblib \
    zmq \
    docker-buildtool

RUN cd /root; \
    git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision; \
    mkdir /root/bullet3/build; \
    cd /root/bullet3/build; \
    cmake -DBUILD_SHARED_LIBS=ON -DBT_USE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=/root/code/roboschool/roboschool/cpp-household/bullet_local_install ..; \
    make -j4 ; \
    make install

```

If you do the same steps, I'll get a working installation (you can skip tini and maybe more). Note that nothing unusual is in there. Can you try it again please so we make the work easier for others?

artsuzy, 2017-06-12:
I have tried to install roboschool through conda `source activate` (env) and I have followed all instructions from readme but it haven't worked

nchentanez, 2017-06-22:
I encountered the same issue but still cannot resolve it. I can run other OpenGL programs and Qt examples fine.

nchentanez, 2017-06-22:
[2017-06-22 22:45:24,873] Making new env: RoboschoolHumanoidFlagrun-v0
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform offscreen surface, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWidget: Failed to create context
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed

nchentanez, 2017-06-22:
Here is my glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: nouveau (0x10de)
    Device: NV120 (0x17c2)
    Version: 12.0.6
    Accelerated: yes
    Video memory: 12271MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: nouveau
OpenGL renderer string: Gallium 0.4 on NV120
OpenGL core profile version string: 3.3 (Core Profile) Mesa 12.0.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_AMD_conservative_depth, GL_AMD_draw_buffers_blend, 
    GL_AMD_seamless_cubemap_per_texture, GL_AMD_shader_trinary_minmax, 
    GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
    GL_ARB_ES2_compatibility, GL_ARB_ES3_compatibility, 
    GL_ARB_arrays_of_arrays, GL_ARB_base_instance, GL_ARB_blend_func_extended, 
    GL_ARB_buffer_storage, GL_ARB_clear_buffer_object, GL_ARB_clear_texture, 
[... 179 lines cut ...]
    GL_EXT_discard_framebuffer, GL_EXT_draw_buffers, 
    GL_EXT_draw_buffers_indexed, GL_EXT_draw_elements_base_vertex, 
    GL_EXT_map_buffer_range, GL_EXT_multi_draw_arrays, 
    GL_EXT_polygon_offset_clamp, GL_EXT_read_format_bgra, 
    GL_EXT_separate_shader_objects, GL_EXT_shader_integer_mix, 
    GL_EXT_texture_border_clamp, GL_EXT_texture_compression_dxt1, 
    GL_EXT_texture_filter_anisotropic, GL_EXT_texture_format_BGRA8888, 
    GL_EXT_texture_rg, GL_EXT_texture_sRGB_decode, 
    GL_EXT_texture_type_2_10_10_10_REV, GL_EXT_unpack_subimage, 
    GL_KHR_context_flush_control, GL_KHR_debug, 
    GL_KHR_robust_buffer_access_behavior, GL_NV_draw_buffers, 
    GL_NV_fbo_color_attachments, GL_NV_read_buffer, GL_NV_read_depth, 
    GL_NV_read_depth_stencil, GL_NV_read_stencil, GL_OES_EGL_image, 
    GL_OES_EGL_image_external, GL_OES_EGL_sync, 
    GL_OES_compressed_ETC1_RGB8_texture, GL_OES_depth24, GL_OES_depth_texture, 
    GL_OES_depth_texture_cube_map, GL_OES_draw_buffers_indexed, 
    GL_OES_draw_elements_base_vertex, GL_OES_element_index_uint, 
    GL_OES_fbo_render_mipmap, GL_OES_get_program_binary, GL_OES_mapbuffer, 
    GL_OES_packed_depth_stencil, GL_OES_rgb8_rgba8, GL_OES_sample_shading, 
    GL_OES_sample_variables, GL_OES_shader_multisample_interpolation, 
    GL_OES_standard_derivatives, GL_OES_stencil8, GL_OES_surfaceless_context, 
    GL_OES_texture_3D, GL_OES_texture_border_clamp, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_texture_stencil8, GL_OES_vertex_array_object

olegklimov, 2017-07-18:
> I encountered the same issue but still cannot resolve it. I can run other OpenGL programs and Qt examples fine. Any help would greatly be appreciated!

Hey @nchentanez any news?

nchentanez, 2017-07-18:
The issue was that I used Anaconda 4.4.0 that came with Python 3.6 and some library didn't support 3.6 so I downgraded Python to 3.5 manually. This somehow causes the error.
However, if I create a separate conda environment with python=3.5, and added from OpenGL import GL, then it works. 
There was also an issue with UEFI that causes display driver installation to fail and the same errors are produced. I ended up forcing installation to non-UEFI as well. Not sure if the UEFI case will work with a separate conda environment though, as I didn't test that.

olegklimov, 2017-07-18:
That still sounds like nothing we can fix.

garymcintire, 2017-07-31:
I also had the same problem and spent a lot of time on it before I found this thread. I had a lot of trouble with assimp and Qt5. This is on ubuntu 16.04 and anaconda python. It would be well worthwhile if the anaconda python 3.6 issue could be identified so we wouldn't have to do this

Here are my full notes. Probably some errors as editing after the fact can cause, but maybe it can help others.

conda create --name python3.5 python=3.5
source activate python3.5
export PKG_CONFIG_PATH=/home/gary/anaconda/envs/python3.5/lib/pkgconfig  
################# gym #############
cd
git clone https://github.com/openai/gym.git
cd gym
sudo apt-get install -y python-numpy python-dev cmake zlib1g-dev libjpeg-dev xvfb libav-tools xorg-dev python-opengl libboost-all-dev libsdl2-dev swig
pip install -e '.[all]'

###### roboschool https://github.com/openai/roboschool
apt install cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev
export ROBOSCHOOL_PATH=/home/gary/roboschool
cd
git clone https://github.com/openai/roboschool
cd roboschool
git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision
mkdir bullet3/build
cd    bullet3/build
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..
make -j4
make install
cd ../..
For Qt5, go here and use the auto installer https://www.qt.io/download-open-source/#section-2  #  it installs 1.5G Yuck!
cd ~/Downloads
sudo ./qt-unified-linux-x64-3.0.0-online.run
conda install boost   # not sure if this is necessary
pip install -e $ROBOSCHOOL_PATH	# pip is my pip3 since I was in python3 environment
THAT WAS IT!
vglrun python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py  # to test

rmst, 2017-07-31:
Thanks @garymcintire , using python 3.5 instead of 3.6 worked for me, too. Just had to do:

conda create --name python3.5 python=3.5
source activate python3.5
pip install gym
pip install -e roboschool

Did **not** need to install https://www.qt.io/download-open-source/#section-2 . But had to use vglrun (would get initGL() Assertion r0 failed, otherwise).

DennisSoemers, 2017-09-19:
For me, this issue seems to have been fixed in Ubuntu 16.04, Anaconda, Python3.6 (so no need to use 3.5) using

    conda install --force -c conda-forge icu qt

Though now I'm getting different errors, described in https://github.com/openai/roboschool/issues/15, so gotta find a fix for that next... :)

ouyangzhuzhu, 2018-11-24:
so sad sad I meet the same problem and add from OpenGL import GLU to the beginning of python main did not work for me............: sad,,,,,,,,,,,, can u help me @olegklimov 

Cannot initialize OpenGL context.
Requested version: 4.1
Actual version: 1.4
(it must be at least 3.3 to work)
For possible fixes, see:

#2

python3: render-simple.cpp:695: void SimpleRender::opengl_init(const boost::shared_ptrHousehold::World&): Assertion `0' failed.
Aborted (core dumped