openai/roboschool #28: composeAndFlush: makeCurrent() failed

Simplex2, 2017-06-13:
I installed as the instructions in ubuntu 16.10, but when I try to run the pretrained agent, I got this message continually. 

`composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
`
What's the problem?

olegklimov, 2017-06-13:
Something similar fixed in #2, please take a look.

olegklimov, 2017-07-18:
Stalled issue, please let me know if there's something we can fix.

closed 2017-07-18 by olegklimov

toothlessdragon, 2018-03-08:
I took a look at #2 but I still didn't see how to fix ```composeAndFlush: makeCurrent() failed```.

I'm on ubuntu 16.04 (azure VM) trying to see the video over X11 forwarding.  I set up my system using this script:

```
#!/bin/sh
sudo apt-get update;
sudo apt-get upgrade -y;
sudo apt-get install -y cmake ffmpeg pkg-config qtbase5-dev libqt5opengl5-dev libassimp-dev libpython3.5-dev libboost-python-dev libtinyxml-dev;
git clone https://github.com/openai/roboschool.git ~/roboschool;
cd ~/roboschool;
ROBOSCHOOL_PATH=$(pwd);
cd ..;
git clone https://github.com/olegklimov/bullet3 -b roboschool_self_collision;
mkdir bullet3/build;
cd bullet3/build;
cmake -DBUILD_SHARED_LIBS=ON -DUSE_DOUBLE_PRECISION=1 -DCMAKE_INSTALL_PREFIX:PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install -DBUILD_CPU_DEMOS=OFF -DBUILD_BULLET2_DEMOS=OFF -DBUILD_EXTRAS=OFF  -DBUILD_UNIT_TESTS=OFF -DBUILD_CLSOCKET=OFF -DBUILD_ENET=OFF -DBUILD_OPENGL3_DEMOS=OFF ..;
make -j4;
make install;
cd ../..;
sudo apt-get install -y python3-pip;
pip3 install --upgrade pip;
sudo -H pip3 install pyopengl;
pip3 install -e $ROBOSCHOOL_PATH;
```

PandaYue99, 2018-05-30:
I am getting the same error message of composeAndFlush:makeCurrent() failed  
I am on Ubuntu 16.04 conda+ gym+tensorflow installed, during installation, no error message occurred.

codydg, 2020-04-29:
Per [this user's comment](https://github.com/openai/roboschool/issues/2#issuecomment-456541588) on another issue, if you're using Windows 10's WSL and XLaunch, Make sure to start XLaunch in the "Single large window" mode, not the default "Multiple Windows" mode. This fixed the issue for me.