openai/roboschool #29: Segmentation fault on env.reset()

jaidmin, 2017-06-13:
Hello, 

I'm trying to get roboschool running on arch and everything seems to work fine while installing, but when I try to run any of the examples or just try to use roboschool environment I get a seg fault:

>>> import gym
>>> import roboschool
>>> env = gym.make("RoboschoolReacher-v0")
[2017-06-13 11:51:27,883] Making new env: RoboschoolReacher-v0
>>> env.reset()
b3Error[/home/jaidmin/Software/Sources/bullet3/examples/SharedMemory/PhysicsServerCommandProcessor.cpp,1904]:
[1]    11654 segmentation fault (core dumped)  python

I am not using Anaconda, I installed all dependencies via the system package manager

When I try to compile bullet itself everything works fine, I can run all the examples and there are no problems

If I can provide any more information please let me know

olegklimov, 2017-06-13:
Please see if #7 is similar to your case.

Phylliade, 2017-06-19:
Hi,

Exactly the same error here, with no other error than the segfault.

Here is the beginning of the stack trace from the coredump:
```
                Stack trace of thread 31785:
                #0  0x00007fa9024142a3 strchrnul (libc.so.6)
                #1  0x00007fa9023d0af2 vfprintf (libc.so.6)
                #2  0x00007fa9023f9700 vsnprintf (libc.so.6)
                #3  0x00007fa8f5b2645e b3OutputErrorMessageVarArgsInternal (libBullet3Common.so.2.86)
                #4  0x00007fa8f4fafbe5 _ZN18BulletMJCFImporter15parseMJCFStringEPKcP15MJCFErrorLogger (libPhysicsClientC_API.so.2.87)
                #5  0x00007fa8f4faeded _ZN18BulletMJCFImporter8loadMJCFEPKcP15MJCFErrorLoggerb (libPhysicsClientC_API.so.2.87)
                #6  0x00007fa8f4f33c69 _ZN29PhysicsServerCommandProcessor8loadMjcfEPKcPcibi (libPhysicsClientC_API.so.2.87)
                #7  0x00007fa8f4f3c8e1 _ZN29PhysicsServerCommandProcessor14processCommandERK19SharedMemoryCommandR18SharedMemoryStatusPci (libPhysicsClientC_API.so.2.87)
                #8  0x00007fa8f4f2ba34 _ZN13PhysicsDirect19submitClientCommandERK19SharedMemoryCommand (libPhysicsClientC_API.so.2.87)
                #9  0x00007fa8f4f602f4 b3SubmitClientCommand (libPhysicsClientC_API.so.2.87)
                #10 0x00007fa8f4f6037f b3SubmitClientCommandAndWaitStatus (libPhysicsClientC_API.so.2.87)
                #11 0x00007fa8f8c5938a _ZN9Household5World13load_sdf_mjcfERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb (cpp_household.so)
                #12 0x00007fa8f8c93622 _ZN5World9load_mjcfERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE (cpp_household.so)
                #13 0x00007fa8f8c8e706 _ZN5boost6python6detail6invokeINS0_15to_python_valueIRKNS0_4listEEEM5WorldFS4_RKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEENS0_15arg_from_pythonIRS8_EENSJ_ISG_EEEEP7_objectNS1_11invoke_tag_ILb0ELb1EEERKT_RT0_RT1_RT2_ (cpp_household.so)
                #14 0x00007fa8f4cabe45 _ZNK5boost6python7objects8function4callEP7_objectS4_ (libboost_python3.so.1.64.0)
                #15 0x00007fa8f4cac008 n/a (libboost_python3.so.1.64.0)
                #16 0x00007fa8f4cb2973 _ZN5boost6python21handle_exception_implENS_9function0IvEE (libboost_python3.so.1.64.0)
                #17 0x00007fa8f4ca9709 n/a (libboost_python3.so.1.64.0)
                #18 0x00007fa901fe414c _PyObject_FastCallDict (libpython3.6m.so.1.0)
```
This happened on Arch, with Mesa + Intel drivers and libglvnd (which is the default on Arch). It's both crashing with Xorg and Wayland.

olegklimov, 2017-06-21:
Never seen this problem as well :(  If it is a stable error that you get 100% of the time, please do the following:

- reconfigure Bullet in debug mode (CMAKE_BUILD_TYPE=DEBUG),
- import cpp_household_d just in case,
- run it under gdb,
- inspect what really crashes it, using gdb commands (bt, up, down, p)

I suspect some file is not found, but error message printing is broken.

wtipton, 2017-06-28:
Hi, 
I'm also on Arch Linux with this problem, and I think the issue is a mismatch of tinyxml.h version in bullet. The following fixed it for me:

On line 2 of bullet3/examples/Importers/ImportMJCFDemo, change
#include "../../ThirdPartyLibs/tinyxml/tinyxml.h"
to
#include "tinyxml.h"

olegklimov, 2017-06-28:
Wow. At least that looks like it! If it will turn out to be true, then you're my hero @wtipton , you save the day!

jaidmin, 2017-06-29:
@wtipton Amazing that fixed it, thanks alot!

Still some problems setting everything up... 
I can run it now using the system python, but with anaconda I get a QT version missmatch because the system version (5.9) is not equal to anacondas version (which will stay at 5.6 see here https://groups.google.com/a/continuum.io/forum/#!topic/anaconda/2OHQXAwHI9I )

any idees how I could force roboschool to use the system QT ?  

This is the exact error:

Traceback (most recent call last):
  File "/home/jaidmin/Software/Sources/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 1, in <module>
    import gym, roboschool
  File "/home/jaidmin/Software/Sources/roboschool/roboschool/__init__.py", line 80, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/home/jaidmin/Software/Sources/roboschool/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/home/jaidmin/Software/Sources/roboschool/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: /home/jaidmin/anaconda2/envs/tensorflow/bin/../lib/libQt5Core.so.5: version `Qt_5.9' not found (required by /home/jaidmin/Software/Sources/roboschool/roboschool/cpp_household.so)

rhofour, 2017-06-30:
It doesn't feel right to depend on anything in bullet3/examples/. Examples should be to demonstrate how the library works and certainly aren't part of the API.

If roboschool needs the example MuJoCo importer then we should copy it into either it's own repo or this one. Directly depending on example code seems like a bad practice.

rhofour, 2017-07-01:
So, in the more short term I looked into what @wtipton found and it doesn't look like we can rely entirely on the tinyxml copy that comes with bullet. I no longer get any segfaults if I simply remove -ltinyxml from roboschool/cpp-household/Makefile. It looks like we can simply remove the requirement on tinyxml and solve this problem with any change to bullet.

olegklimov, 2017-07-18:
Removed -ltinyxml

Let's see how that goes! Please test v1.1

closed 2017-07-18 by olegklimov

Phylliade, 2017-07-18:
I can confirm it's working!

rdaries29, 2017-07-22:
Hi,

I seem to be getting two forms of errors. 1.) A segmentation error when I attempt to execute a demo example 2.) A missing image when I attempt to import roboschool. I have successfully installed roboschool. Is there anything I am not seeing?

[image]

[image]