openai/roboschool #31: ImportError: dlopen(/Users/MufeiLi/roboschool-master/roboschool/cpp_household.so, 2): Library not loaded: @rpath/libLinearMath.2.87.dylib

mufeili, 2017-06-20:
I got this error when trying to run something from the agent zoo after following the installation instructions on Mac, homebrew python.

> >>> import roboschool
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/MufeiLi/roboschool-master/roboschool/__init__.py", line 80, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/Users/MufeiLi/roboschool-master/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/Users/MufeiLi/roboschool-master/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: dlopen(/Users/MufeiLi/roboschool-master/roboschool/cpp_household.so, 2): Library not loaded: @rpath/libLinearMath.2.87.dylib
  Referenced from: /Users/MufeiLi/roboschool-master/roboschool/cpp_household.so
  Reason: image not found

I've searched among the issues and this problem seems very similar to #10  . In fact, I managed to get the demo to work by 

> export LD_LIBRARY_PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install/lib/

Is this normal? Is there a way to fix it? Thanks.

olegklimov, 2017-06-21:
> Is this normal

No it's not normal. Don't know how to reproduce it though, works for me. :(

rdaries29, 2017-07-25:
Hi, I seem to get the same error as @mufeili . Once I added the `export` command I then got a segmentation error.

[image]

[image]

olegklimov, 2017-07-25:
Yep. If you can fix it, post a fix for everyone else.

joshwang0322, 2017-08-05:
I've encountered the same issue. I can call python to run demo in terminal. But if I start ipython notebook via Anaconda, it always gave me the "cpp_household.so library not loaded" error when trying to import roboschool.

Not sure if you've already found this, but adding these lines into .bash_proflie did help:
`#roboschool`
`export ROBOSCHOOL_PATH=/Users/xingyuwang/roboschool-master`
`export LD_LIBRARY_PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install/lib/`
(for ROBOSCHOOL_PATH I'm using my local path of roboschool project)

After this I could import roboschool in Jupyter notebook....

mrkent, 2017-08-06:
Having the same problem, even after adding `LD_LIBRARY_PATH`.

```
export ROBOSCHOOL_PATH=/Users/mrkent/workspace/roboschool
export LD_LIBRARY_PATH=$ROBOSCHOOL_PATH/roboschool/cpp-household/bullet_local_install/lib/
```

```
mrkent:~$ /usr/local/bin/python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
Traceback (most recent call last):
  File "/Users/mrkent/workspace/nn/cs294/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 1, in <module>
    import gym, roboschool
  File "/Users/mrkent/workspace/nn/cs294/roboschool/roboschool/__init__.py", line 92, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/Users/mrkent/workspace/nn/cs294/roboschool/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/Users/mrkent/workspace/nn/cs294/roboschool/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: dlopen(/Users/mrkent/workspace/nn/cs294/roboschool/roboschool/cpp_household.so, 2): Library not loaded: @rpath/libQt5OpenGL.5.dylib
  Referenced from: /Users/mrkent/workspace/nn/cs294/roboschool/roboschool/cpp_household.so
  Reason: image not found

```

ShangtongZhang, 2017-10-27:
I met same issue as mrkent.. don't know how to solve it

ShangtongZhang, 2017-10-27:
I'm using os x 10.12, I find it only happens with anaconda. If I use homebrew python, it works well.