openai/roboschool #32: Training HumanoidFlagrunHarder

ViktorM, 2017-06-21:
Hi,

May be I missed the related detail in blog article or arxiv paper, but can you to shed some light on how  the HumanoidFlagrunHarder was trained? What algorithm was used, how the training was organised?

olegklimov, 2017-06-21:
Harder versions never made it into any paper, probably will though.

Algorithm is unreleased yet (it will be), but algorithm is not that important. Any will do, maybe with less effectiveness but many of them should work.

olegklimov, 2017-07-18:
PPO paper is going out today (or maybe tomorrow).

closed 2017-07-18 by olegklimov

ViktorM, 2017-07-18:
@olegklimov, what a name of the paper? The only PPO paper that I've seen so far was Deepmind's "Emergence of Locomotion Behaviours in Rich Environments" but it has been placed to the arxiv a few days before your reply and it had nothing to do with the Humanoid Hard Flagrun.

olegklimov, 2017-07-19:
Yeah they use it too. It was unpublished, but mentioned numerous times in tutorials and workshops. They even correctly mention it, props to them.

ViktorM, 2017-07-20:
I've seen recent blog post on OpenAI site and baselines TRPO and PPO addition. Great work!