openai/roboschool #33: Readme install instructions for Mac

curtwelch, 2017-06-21:
Just wanted to share my problems with the developers.  I just installed roboschool on my Mac as part of the Udacity Nanodegree and ran into many issues due to my lack of understanding of these sorts of packages.

1) 

The install instructions say First define a ROBOSCHOOL_PATH variable.

In fact, step one (First) should be to git clone the repo to the local computer.  That might be assumed by many, but since I was just trying to follow the instructions I didn't understand that, I just created a roboschool directory to do the installs in and followed the instructions.  Then, when pip didn't work (I don't know what pip is or how it works or how you use it -- I was just following the instructions).  After research, it became obvious I should have STARTED by git cloning the repository.  But you can't do it after the other installs since they try to do the build inside the repo that it assumes you already installed.  So I had to move stuff around, clean it all up, and start over.

Notice how openAI gym has install instructions like this:

git clone https://github.com/openai/gym.git
cd gym
pip install -e .

It would help us that are novices at all that to make the git clone the obvious first step of the install, and also make it obvious what the PATH needs to be defined as (because it gets very confusing later when you create a roboschool directory in the roboschool directory making it look like a path is mis-configred.

So maybe something like this would be good as step one:

git clone https://github.com/openai/roboschool.git
cd roboschool
ROBOSCHOOL_PATH=`pwd`

2)
I used the "Mac, Anaconda with Python 3" instructions since we had set up Anaconda in the course (again, all brand new to me).

I didn't have homebrew installed, and I've never HEARD of homebrew so I had to go to google for help and followed some random instructions found on some random forum.  I have no clue if I actually installed the right version of brew the correct way or not.

Maybe a link to where you should go to install homebrew would be nice?

3)

The pip3 install of roboschool first failed because I didn't have pkg-info installed on my Mac.  Again, it would be nice to include that in the step by step commands (You did mention it in the requirements).

I just used "brew install pkg-config" but again, I don't know if that was correct. But it seemed to work.

4)

Trying to run the roboschool code failed on "import gym".

What's gym?  I've never heard of it and you never mentioned it in the instructions?  Every try to google for "gym"? :)

Yes, the context is all very clear to YOU since this is a project connected with OpenAI GYM, but I came to your project from a link that did not explain what this project was about.

It would be nice if in the install instructions, you mention, Oh, BTW, you have to install https://github.com/openai/gym as well for those of us lost souls trying to check out your work!

Thanks, just wanted to share some of my frustrations for the day that could be improved by a few updates to the install instructions in the readme file!

Thanks, Curt

olegklimov, 2017-06-21:
Did it start to work in the end?

curtwelch, 2017-06-22:
Yes, thanks, I did get it installed and running!  I think it's working
correctly but I'll have to spend more time with it to know for sure if it's
working as it should.  The apps run and a graphics are displayed so it
looks correct.

Curt

Jdgmx, 2017-06-22:
Thank you very much for this Curt.
I also had to install roboschool for the Udacity program and went through all the problems that you mention and wasted almost a complete week of time just for this.

But for me the pain is not over, finally I get to the point of installing roboschool but I get this error:

$ pip3 install -e $ROBOSCHOOL_PATH
Obtaining file:///Volumes/Essd/SourceBase/robots/roboschool
Installing collected packages: roboschool
  Found existing installation: roboschool 1.0
    Uninstalling roboschool-1.0:
      Successfully uninstalled roboschool-1.0
  Running setup.py develop for roboschool
Successfully installed roboschool

$ python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
Traceback (most recent call last):
  File "/Volumes/Essd/SourceBase/robots/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 1, in <module>
    import gym, roboschool
ModuleNotFoundError: No module named 'roboschool'

Successfully installed roboschool but no module named 'roboschool' ?
Now I'm again at the point were I don'y know what to do.
I'll appreciate some help here.

- Joaquin.

olegklimov, 2017-06-22:
Try `python3`.

Jdgmx, 2017-06-22:
$ python3 $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
Traceback (most recent call last):
  File "/Volumes/Essd/SourceBase/robots/roboschool/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 1, in <module>
    import gym, roboschool
ModuleNotFoundError: No module named 'roboschool'

Same thing :(

olegklimov, 2017-06-22:
Not a lot can go wrong there. Just look for one python3 it got installed into, another python3 you're running. Try installing other random packages to make sure your system works at all.

curtwelch, 2017-06-22:
Your problem is an incompatibility between the pip you are using and the
Python you are using.  Pip installed roboschool SOMEWHERE on your system,
but the version of python you ran, looked in a different place for the
installed package and didn't find it.

Like Oleg said, the most likely answer is if you run pip3 to do the
install, you need to run python3 so it looks in the same place pip3 put the
package.

but someone (was it you) on the Udacity slack, said using python3 didn't
make the error go away.  This would indicate you have some sort of strange
messed up installation where the pip3 and python3 you are using are not a
correctly matched pair.

If you are using conda (anaconda), it changes where pip and Python stores
packages.  So you might be using the conda version of either pip or python
but not the correct conda version of the other???  I don't fully understand
all the complexity here.

But, one thing to try, if python3 doesn't fix this, (I see your new email
now and you say it doesn't), then try installing with "pip" instead of
pip3.  See if that makes it work with either python or python3.

Curt

Jdgmx, 2017-06-22:
Hello. Thank you very much for your help Yes, the problem is that my python installation is all messed up. I have python 3 installed twice, one by the system (I guess) and another by miniconda3.

olegklimov, 2017-07-18:
Please let me know if there's something to fix.

closed 2017-07-18 by olegklimov