openai/roboschool #36: Fedora support

rhofour, 2017-06-26:
As far as I can see there are only two things missing to support Fedora installation of Roboschool.

1) List of dependencies. I was able to install it with: dnf install cmake ffmpeg pkgconfig qt5-devel assimp-devel python3-devel boost-python3-devel tinyxml-devel

2) Modification to cpp-household/Makefile. Right now the Makefile treats all Linux environments like Ubuntu. The one place I found this problematic was linking to boost-python for Python 3.5. On Ubuntu this is with boost-python-py35, but on Fedora this is just boost-python. I'm not terribly familiar with Make so I don't know the best way to fix this.

rhofour, 2017-06-26:
Actually, that may not be all I need. I got everything installed, but when I try and run anything from the model zoo I get: ImportError: <MY PATH>/roboschool/roboschool/cpp_household.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE

I'm not sure if this is a Roboschool problem or a problem with my system.

olegklimov, 2017-06-26:
I guess you've linked against boost-python for python2.

(The real solution to build problems would be to use proper build system, that already knows or autodetects all libraries, works with pip, but I don't know such system :( )

rhofour, 2017-06-26:
Just figured that out. Now I get a new complaint instead:

```
[2017-06-25 22:20:18,167] Making new env: RoboschoolHumanoidFlagrun-v0
b3Warning[/home/rofer/Dropbox/Apps/roboschool-bullet3/examples/SharedMemory/PhysicsServerCommandProcessor.cpp,1908]:
/home/rofer/Dropbox/Apps/roboschool/roboschool/mujoco_assets/ground_plane.xml:3: unhandled root element ''b3Error[/home/rofer/Dropbox/Apps/roboschool-bullet3/examples/SharedMemory/PhysicsServerCommandProcessor.cpp,1904]:
Segmentation fault (core dumped)
```

olegklimov, 2017-06-26:
Similar to #29  (reason unknown)

rhofour, 2017-06-26:
Ah, thanks! It's perfectly reproducible for me so I'll dig into a bit further tomorrow.

rhofour, 2017-06-27:
Actually, after looking into it, I seem to be seeing something a bit different since I can't even run the examples.

After running things in GDB I see the problem appears to be with the XML parsing. In BulletMJCFImporter::parseMJCFString xml_doc is returning error 32 and when ErrorDesc is called it returns 0x10 rather than a valid pointer to a string. This of course causes a segfault when it's dereferenced.

Not sure if this is related to XML or if the bug is entirely in the parser. The XML it appears to be reading is:
> "<mujoco model=\"ground_plane\">\n  <worldbody>\n    <geom conaffinity=\"1\" condim=\"3\" name=\"floor\" friction=\"0.8 0.1 0.1\" pos=\"0 0 0\" type=\"plane\"/>\n  </worldbody>\n</mujoco>\n\n"

olegklimov, 2017-07-18:
Closed #29 , maybe this will go away, too.

closed 2017-07-18 by olegklimov