openai/roboschool #39: Changing parameters of environment

anaypat, 2017-07-08:
Hi,
Is there any way I can change parameter of dynamics for environment? For example, how can can I change the mass of torso of hopper?

olegklimov, 2017-07-08:
That is a good idea. For now, you have to generate random .xml and load that. Feel free to share if you have good implementation.

anaypat, 2017-07-08:
I see that default parameter files (.xml) get loaded through mujoco assets. Is there any way to integrate this with mujoco-py (https://github.com/openai/mujoco-py) binding? I've changed parameters of env before through mujoco-py

anaypat, 2017-07-08:
I guess it can be done through Mujoco-py. Will give it a try and let you know how it goes

olegklimov, 2017-07-18:
This project doesn't use mujoco. Let alone mujoco-py.

anaypat, 2017-07-18:
I could use mujoco for simulations. Unfortunately, I wasn't able to use mujoco-py with it. Mujoco-py would have been great as I can change parameters easily with it.

anaypat, 2017-07-18:
Sorry, let me take back my statement, I got it working with rllab, not mujoco. Instead of changing parameters by editing the xml files explicitly, I tried to use mujoco-py for making changes just like it does to the xml files for mujoco. Unfortunately, it didn't work out.

whoiscris, 2019-06-03:
Is it possible now also in roboschool to change the .xml while running the env(and not having to reload the env) ?