openai/roboschool #41: Bug: `AssertionError` in `gym_pendulums.py`: slider position x is not finite

denizs, 2017-07-17:
# Expected behaviour
You can use the openai `gym` API to interact with the `RoboschoolInvertedPendulumSwingup-v0` environment

# Observed behaviour
Calling `step(env.action_space.sample())` throws an `AssertionError`, because the `x` slider position is not finite.

I will dig deeper into this later and try to fix it. In the meantime - is there anyone who can reproduce this?

```python
import roboschool
import gym

env = gym.make('RoboschoolInvertedPendulumSwingup-v0')
env.reset()
env.step(env.action_space.sample())  # <-- throws the `AssertionError` in `calc_state()`
```

olegklimov, 2017-07-17:
>  is there anyone who can reproduce this

I've ran your example, it works for me. Are you even asking a question or providing an example?

Observations are infinite most likely because actions are infinite.

Closing for now (bug totally unclear).

closed 2017-07-17 by olegklimov

denizs, 2017-07-17:
Ok sorry for my poor description.
What I meant was that calling `step` on the `env` instance raised an `AssertionError` located in `roboschool/gym_pendulums.py` on line 85.

After restarting my machine, I couldn't reproduce the behaviour. After fiddeling around for some time, I figured out, that the actual reason is calling `reset()` when `done` is not `True`.

When I do the following, the env is stuck and can't recover. When calling `step`, `state = self.calc_state()` on line 93 throws the `AssertionError`, calling `reset()` again and retrying does not solve the issue.

I'm aware, that this is a real edge case, as you normally don't call `reset()` on a non terminal state, so you can keep this closed.
Thanks for your time though!

```python
import roboschool
import gym

env = gym.make('RoboschoolInvertedPendulumSwingup-v0')
env.reset()
env.render(mode='rgb_array')
env.step(env.action_space.sample())
env.reset()
env.render(mode='rgb_array')
env.step(env.action_space.sample()) # <-- throws the `AssertionError` in `calc_state()`
```

olegklimov, 2017-07-18:
Still works for me. Even in a loop. With or without render.

denizs, 2017-07-18:
That's odd. I'm able to reproduce this regardless of the render. 
Thanks for looking into it though!