openai/roboschool #42: Qt headers missing on Ubuntu 14.04

danijar, 2017-07-22:
Output from `pip3 install -e $ROBOSCHOOL_PATH`:

```
In file included from render-glwidget.h:1:0,

                 from render-glwidget.cpp:2:

render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory

 #include <QtWidgets/QOpenGLWidget>

                                   ^

compilation terminated.
```

Output from `sudo apt-get install qtbase5-dev libqt5opengl5-dev`:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt5opengl5-dev is already the newest version (5.2.1+dfsg-1ubuntu14.3).
qtbase5-dev is already the newest version (5.2.1+dfsg-1ubuntu14.3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

olegklimov, 2017-07-22:
Hi. Do you think you can fix it?

danijar, 2017-07-22:
@olegklimov I'd be happy to help fixing it but I have no idea how to go from here. Do you have any idea?

olegklimov, 2017-07-22:
Yes. I find that folder, that file, path it searches for, match it, find out what's wrong.

I can't actually do that, because I don't have 14.04 and I have another work to do :-(  Of course I wish I had 42 hours a day ...

danijar, 2017-07-22:
It's trying to include the header file of Qt's OpenGL extension. That would indicate that either Qt or Qt's OpenGL extension is not installed. But they are, as shown in the first post. It could be the case that the paths are not visible for some reason, or that Ubuntu 14.04's repositories contain an incompatible version of the libraries that names its headers files differently.

olegklimov, 2017-07-22:
Headers are the same 100%. Qt 5 is ages old.

danijar, 2017-07-24:
@olegklimov Do you have an idea how to further debug this problem?

chaotaklon, 2017-07-26:
I use Centos so I don't use apt-get to install Qt, instead, I download the .run file from Qt official site.
Run it, there will be a GUI for you to install. Then, Qt will be in /opt/Qt
By the way, pip doesn't work but I don't know why. Only pip3 works.
Before pip3 it, you may need to run:
$ export C_INCLUDE_PATH=/usr/local/include/assimp:/opt/Qt/5.9.1/gcc_64/include
$ export CPLUS_INCLUDE_PATH=/usr/local/include/assimp:/opt/Qt/5.9.1/gcc_64/include
$ export LIBRARY_PATH=/usr/local/lib/pkgconfig/:/opt/Qt/5.9.1/gcc_64/lib/pkgconfig
$ export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig/:/opt/Qt/5.9.1/gcc_64/lib/pkgconfig

olegklimov, 2017-07-26:
Correct. Maybe you just need `PKG_CONFIG_PATH`, other paths will be found by pkg-config.

navigator8972, 2017-08-02:
The problem seems related to the header "QtWidgets/QOpenGLWidget", as @danijar shows. This file has been available in Qt5 since Qt5.4 (http://doc.qt.io/qt-5/qopenglwidget.html), but the default Qt5 package under 14.04 Trusty is 5.2.1(https://packages.ubuntu.com/trusty/qt5-default). The solution from @chaotaklon worked, although I used the Qt5.7 which was installed before. Confirm that only <code>PKG_CONFIG_PATH</code> is needed.

I also found a similar issue about assimp, which is v3.0 under Trusty. It seems that we need a higher version (v3.3 in my case) to have the macro <code>AI_CONFIG_PP_PTV_ADD_ROOT_TRANSFORMATION</code> defined so the compilation can go.

Also remember to add the lib folders of Qt5 and assimp to <code>LD_LIBRARY_PATH</code> for running the race and humanoid demos. My tensorflow builds on 2.7, so I only tried pip and it worked for me.

kkjh0723, 2017-11-08:
@navigator8972, @olegklimov My system is Ubuntu 14.04, Anaconda and python3.5, and I got the error as below. How can I solve the problem? I read some of the previous issues but couldn't figure out the solution.

    Obtaining file:///home/jinhyung/git/roboschool
        Complete output from command python setup.py egg_info:
        found resource files: 185
        running egg_info
        Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
        Link against python 3.5
        Slow hardware or software render (no shadows)
        rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
        rm -rf .generated
        rm -rf .build-debug
        rm -rf .build-release
        Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
        Link against python 3.5
        Slow hardware or software render (no shadows)
        mkdir -p .generated
        mkdir -p .build-release
        mkdir -p .build-debug
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC     -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
        In file included from render-glwidget.h:1:0,
                         from physics-bullet.cpp:1:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from assets-mesh.cpp:1:0:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from random-world-tools.cpp:3:0:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from render-glwidget.h:1:0,
                         from render-glwidget.cpp:2:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        make: *** [.build-release/physics-bullet.o] Error 1
        make: *** Waiting for unfinished jobs....
        make: *** [.build-release/assets-mesh.o] Error 1
        make: *** [.build-release/random-world-tools.o] Error 1
        make: *** [.build-release/render-glwidget.o] Error 1
[... cut ...]
        Install dependencies that pkg-config cannot find.
        
    
        ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /home/jinhyung/git/roboschool/

Following is `qmake -v` output.

    QMake version 3.0
    Using Qt version 5.6.2 in /home/jinhyung/anaconda3/envs/py35/lib

And `PKG_CONFIG_PATH` is
    
    /home/jinhyung/anaconda3/envs/py35/lib/pkgconfig

And `LD_LIBRARY_PATH` is
/home/jinhyung/anaconda3/envs/py35/lib:~/Qt5.7.0/5.7/gcc_64/lib

zxecho, 2018-01-16:
@kkjh0723 I meet the same problem, have you solved it already? If solved, please reply me, thks. :)

hari-sikchi, 2018-02-10:
Installing Qt from runfile, [Qt Runfile Installation](https://wiki.qt.io/Install_Qt_5_on_Ubuntu) and setting the PKG_CONFIG_PATH solved it for me.
$ export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig/:/home/user/Qt/5.9.1/gcc_64/lib/pkgconfig

zmonoid, 2018-06-02:
consider changing lines 50:54 in roboschool/cpp-household/Makefile
```
$(info Link against python $(PYTHON))
 INC     += `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp`
 LIBS    += -lstdc++ `$(PKG) --libs Qt5OpenGL Qt5Widgets assimp`
 INC     += -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -I/usr/include/python3.5m
 LIBS    += $(RPATH) -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API 
```

fw93, 2018-06-17:
@zmonoid That worked like a charm

novinsh, 2018-09-15:
> consider changing lines 50:54 in roboschool/cpp-household/Makefile to the following
> 
> ```
> $(info Link against python $(PYTHON))
>  INC     += `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp`
>  LIBS    += -lstdc++ `$(PKG) --libs Qt5OpenGL Qt5Widgets assimp`
>  INC     += -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -I/usr/include/python3.5m
>  LIBS    += $(RPATH) -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API 
> ```

@zmonoid This worked for me too. Thanks! 
It seems the only difference with the original block of code was python-$(PYTHON). But what is wrong with this? what does this do?