openai/roboschool #47: Command "python setup.py egg_info" failed with error code 1 in ~/roboschool/

glazari, 2017-08-01:
My installation of roboschool has failed with the following message:
`Obtaining file:///home/glazari/repos/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
    Link against python 3.6
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
    Link against python 3.6
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    In file included from render-glwidget.h:1:0,
                     from physics-bullet.cpp:1:
    render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
    compilation terminated.
    Makefile:130: recipe for target '.build-release/physics-bullet.o' failed
    make: *** [.build-release/physics-bullet.o] Error 1
    make: *** Waiting for unfinished jobs....
    In file included from assets-mesh.cpp:1:0:
    render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
    compilation terminated.
    Makefile:130: recipe for target '.build-release/assets-mesh.o' failed
    make: *** [.build-release/assets-mesh.o] Error 1
    In file included from random-world-tools.cpp:3:0:
    render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
    compilation terminated.
    Makefile:130: recipe for target '.build-release/random-world-tools.o' failed
    make: *** [.build-release/random-world-tools.o] Error 1
    In file included from render-glwidget.h:1:0,
                     from render-glwidget.cpp:2:
    render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
    compilation terminated.
    Makefile:130: recipe for target '.build-release/render-glwidget.o' failed
    make: *** [.build-release/render-glwidget.o] Error 1
    cd /home/glazari/repos/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/glazari/repos/roboschool/
`

Does anyone know what could have caused this?

Lampogolovii, 2017-08-08:
have the same problem... did you resolve it?

nemanja-rakicevic, 2017-08-13:
I also have the same issue. I'm using MacOS Sierra 10.12.6.
I think it's something to do with the pkg-config and the path, because it doesn't find Qt properly
`render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
compilation terminated.`

Bhargav5, 2017-08-28:
I have faced similar issue in Ubuntu 16.04,
I have solved it by changing the gcc version.
You can install gcc 6.1.1 using the instructions mentioned at 
[image]
Then set gcc version to 6.1.1 and use command
sudo -H pip install -e $ROBOSCHOL_PATH

urbanintell, 2017-09-15:
I am on a MAC running macOS Sierra and I have run into this problem. Has anyone found a solution?

panda1221, 2017-09-21:
I am on a MAC running macOS Sierra and I have run into this problem. Has anyone found a solution?

olegklimov, 2017-09-21:
> Package python-3.6 was not found in the pkg-config search path.

Original poster problem is most likely trivial, python-dev needs to be installed. Let's close this.

closed 2017-09-21 by olegklimov

FrancoisPl, 2017-09-30:
I don't know why @Bhargav5 's answer didn't work for me: my gcc version was the one set by anaconda, but it ended up working by adding -DCMAKE_C_COMPILER=/usr/bin/gcc to the cmake options.

stephenjhansen, 2017-10-02:
@olegklimov What is python-dev? I believe I have the same issue and I can't seem to find python-dev. I know this might be trivial but Im having a hard time getting it setup. Thanks.

olegklimov, 2017-10-02:
Python has header files and everything needed to link against python libraries. Just find it in your distro.

sivaram10, 2017-11-26:
I'm facing this same issue.

OS - Ubuntu 16.04
Anaconda 

I did install python-dev, python3-dev and also updated gcc to latest version. I'm still facing this issue. Can someone help me here if you have already resolved this.

TIA

glazari, 2017-11-26:
I am still having the same problem, I installed python-dev, also changes gcc version, but still have the same issue. I am using  a conda virtual environment

Bhargav5, 2017-11-27:
Make sure you are using **sudo -H pip install -e $ROBOSCHOL_PATH** command for installation not only pip install.

sumethy, 2018-05-30:
Activate your conda env then do

`export PKG_CONFIG_PATH=$(dirname $(dirname $(which python)))/lib/pkgconfig`

then `pip3 install -e $ROBOSCHOOL_PATH`

worked for me on Ubuntu 16.04 in a  Python 3.6 conda environment

benvision1, 2018-09-17:
@sumethy, You a life saver. Thanks

psxz, 2018-10-11:
@sumethy, Sounds like it shoud do the trick. Doesn't work for me unfortunately