openai/roboschool #49: Command "python setup.py egg_info" failed with error code 1

acarraro, 2017-08-02:
After I installed everything i run the command 
`pip3 install -e $ROBOSCHOOL_PATH`
and I get the message "Please follow instructions in README to build local (not global for your system) Bullet installation."

Inspecting the source file "https://github.com/openai/roboschool/blob/master/setup.py" I see that the script looks for the subfolder "/roboschool/cpp-household/bullet_local_install/lib" inside the current directory. However I saw that an installation of the bullet engine (http://bulletphysics.org/mediawiki-1.5.8/index.php/Installation) does not create the subfolder "lib". How can I fix this?

olegklimov, 2017-08-02:
I believe there is nothing to fix -- just try again installing Bullet from a branch.

closed 2017-08-02 by olegklimov

Anjum48, 2017-10-31:
I'm getting the same issue. `bullet_local_install/lib` is not being created if I install bullet using the instructions in the README