openai/roboschool #52: Use multiple camera views together with world.test_window

navigator8972, 2017-08-09:
I am trying to use multiple camera windows together with the main Viz. Is it currently supported?

I failed to do so by adapting the [Wiki](https://github.com/openai/roboschool/wiki/Test-Window-and-Camera-Test-Window). Generally, the issues I encountered are:
1. cannot call render("human") and camera.render() (or mycam.render()) in the same loop. Doing this will raise "Assertion e == 0" failure in camera_render() in `render-glwidget.cpp`.
2. the program crashes when render("human") and camera.test_window() (no render called before) are called in the same loop.
3. however, it runs when render("human") and mycam.test_window() if created_camera.render() is not called.
4. calling camera.render() and mycam.render() in the same loop seems okay, although the test_window has no response.

Here is a minimal example to reproduce the findings
```python
import os.path, gym
import numpy as np
import roboschool

def demo_run():
    env = gym.make("RoboschoolAnt-v1")

    while 1:
        frame = 0
        score = 0
        restart_delay = 0
        obs = env.reset()

        #create another camera, the behavior is different from the built-in camera of the robot
        #I dont understand why because both of them seem to be created from new_camera_free_float
        mycam = env.unwrapped.scene.cpp_world.new_camera_free_float(320, 200, "my_camera")
        mycam.move_and_look_at(1,1,1, 0,0,0)

        while 1:
            a = np.zeros(env.action_space.shape)
            obs, r, done, _ = env.step(a)

            # world test_window, cannot work with camera, but can go with mycam if mycam.render is not called, while it outputs garbage pixels as the render is not called
            still_open = env.render("human")

            # rgb_array camera, cannot work
            # img = env.render('rgb_array')
            # env.unwrapped.camera.test_window()

            # mycam: cannot work with world test_window if render is called: Assertion `e == 0' failed in camera_render()
            # rgb, depth, depth_mask, labeling, labeling_mask = mycam.render(True, True, True)
            # mycam.test_window()

if __name__=="__main__":
    demo_run()
```

The test was carried out under Ubuntu 14.04, python 2.7 and Qt5.7. Not sure if these are the reasons.

My current workaround is to plot the render outputs in a third-party GUI, just like what `demo_keyboard_humanoid2.py` does. Any ideas about using render("human") and camera.test_window()?

olegklimov, 2017-08-09:
Oh, you have figured out a lot of this by yourself. That's great. Here I modified your example, this version works (on my computer).

```python
import os.path, gym
import numpy as np
import roboschool

def demo_run():
    env = gym.make("RoboschoolAnt-v1")

    while 1:
        frame = 0
        score = 0
        restart_delay = 0
        obs = env.reset()

        #create another camera, the behavior is different from the built-in camera of the robot
        #I dont understand why because both of them seem to be created from new_camera_free_float
        mycam = env.unwrapped.scene.cpp_world.new_camera_free_float(320, 200, "my_camera")
        mycam.move_and_look_at(1,1,1, 0,0,0)

        while 1:
            a = np.zeros(env.action_space.shape)
            obs, r, done, _ = env.step(a)

            # world test_window, cannot work with camera, but can go with mycam if mycam.render is not called, while it outputs garbage pixels as the render is not called
            still_open = env.render("human")

            # rgb_array camera, cannot work
            img = env.render('rgb_array')
            env.unwrapped.camera.test_window()

            rgb, _, _, _, _ = mycam.render(False, False, False)
            mycam.test_window()

if __name__=="__main__":
    demo_run()

```

closed 2017-08-09 by olegklimov

navigator8972, 2017-08-10:
Thanks. However, it keeps giving me the error on my machine as the observaion 1:
```
render-glwidget.cpp:56 ERROR: 0x502
python: render-glwidget.cpp:56: void Household::Camera::camera_render(const boost::shared_ptr<SimpleRender::Context>&, bool, bool, bool): Assertion `e == 0' failed.
Aborted (core dumped)
```
I think it might be due to the dependencies given that it works on your side. Mind sharing the versions of your Qt, OpenGL and Python interpreter?

olegklimov, 2017-08-10:
Have no idea. Maybe you out of video memory for additional buffers? Try making sizes smaller. I mean, nothing in principle prevents you from having as many cameras as you want.

navigator8972, 2017-08-16:
Thanks. I tried to use a 5x5 camera and upgrade everything to 16.04, but with no luck. :-(

That said, I managed to use multiple cameras without the main Viz. The reason I failed to do so before is that ```app->process_events()``` is only called from ```World::test_window()``` and ```World::step()```. Thus the camera paint events were not processed. Adding ```app->process_events()``` to ```Camera::test_window()``` shows two cameras when ```still_open = env.render("human")``` is commented out.

I'm not familiar with the QOpenGL thing but here is what I've found by digging the code a bit for the crash involving Viz and camera: 

adding a ```cx->glcx->makeCurrent(cx->surf)``` in ```Viz::paintGL()``` before ```_render_on_correctly_set_up_context()``` removes the crash with all of the three windows shown. 

The main Viz, of course, shows nothing but a black background because an offscreen surface is assigned. So I'm wondering if the raised OpenGL error is due to an incorrect surface. The camera switches the rendering to an offscreen surface in ```Camera::camera_render()``` and paints an image obtained from the frame buffer. The Viz seems to paint in a different way. Not sure if the surface needs to be switched back before ```Viz::paintGL()```. Any idea about this @olegklimov?

navigator8972, 2017-08-22:
Problem solved by installing nvidia driver and compiling with ```USE_SSAO```.