openai/roboschool #53: Roboschool interfers with pytorch.multiprocessing

AlexPiche, 2017-08-17:
Hi,

I've been trying to implement A3C on the roboschool environment with the pytorch library, but I am getting an error as soon as I import the library. Here's a simpler version of my code:

https://gist.github.com/AlexPiche/5f3dcd2d7129a4454854773e29da541c

and the error that I get:

https://gist.github.com/AlexPiche/02a2571dc3ec047a880b8931468ec3c4

Note that the function train(args, 0) on line 30 works fine and that simply commenting out the 7th line fixes the problem for the multiprocessing call on line 33. 

Python version:

2017-08-17T10:08:28-04:00

CPython 3.6.2
IPython 6.1.0

compiler   : GCC 4.2.1 Compatible Apple LLVM 8.1.0 (clang-802.0.42)
system     : Darwin
release    : 16.7.0
machine    : x86_64
processor  : i386
CPU cores  : 4
interpreter: 64bit

Thank you for your help.

Best,

Alex

olegklimov, 2017-08-17:
I see nothing relevant in your stack. Some random error, I'm afraid you have to dig more.

closed 2017-08-17 by olegklimov

AlexPiche, 2017-08-18:
Hi,

Thank you for taking the time to answer.

If someone runs into the same problem, it looks like it is related to the BLAS libraries used by numpy.dot. One way to fix it is by calling mp.set_start_method('spawn').

source: https://stackoverflow.com/questions/23963997/python-child-process-crashes-on-numpy-dot-if-pyside-is-imported