openai/roboschool #55: env.reset() breaks on the ant robot example

navigator8972, 2017-08-23:
Hi, there. I'm trying to have a keyboard control to reset the environment. I encountered the similar issue described in the follow-up discussion in [issue 7](https://github.com/openai/roboschool/issues/7). Basically, my robot (Baxter, is it too big?) initializes all the joints at zero position after the second reset. And even ```reset_current_position``` and ```set_servo_target``` fail to move the robot, although ```World:bullet_step()``` seems okay as I've checked with:
```cpp
	b3SharedMemoryCommandHandle cmd = b3InitStepSimulationCommand(client);
	
	b3SharedMemoryStatusHandle statusHandle = b3SubmitClientCommandAndWaitStatus(client, cmd);
	int statusType = b3GetStatusType(statusHandle);
	if (statusType != CMD_STEP_FORWARD_SIMULATION_COMPLETED) fprintf(stdout, "Failed to take a simulation step.\n");
```
Running MuJuCo models have a similar problem, though initialized values are infinity. Here is a small example:
```python
import os.path, gym
import numpy as np
import roboschool

from OpenGL import GL

class TestKeyboardControl:
    def __init__(self):
        self.keys = {}
        self.reset = 0

    def key(self, event_type, key, modifiers):
        self.keys[key] = +1 if event_type==6 else 0
        # print ("event_type", event_type, "key", key, "modifiers", modifiers)
        # case insensitive
        if self.keys.get(ord('R'), 0) > 0:
            self.reset = 1

def demo_run():
    env = gym.make("RoboschoolAnt-v1")

    ctrl = TestKeyboardControl()
    cb_set = False

    while 1:
        frame = 0
        score = 0
        restart_delay = 0

        obs = env.reset()

        if not cb_set:
            env.unwrapped.scene.cpp_world.set_key_callback(ctrl.key)
            cb_set = True

        while 1:
            a = np.zeros(env.action_space.shape)
            obs, r, done, _ = env.step(a)

            j = np.array([j.current_position() for j in env.unwrapped.ordered_joints], dtype=np.float32).flatten()
            print(j)

            still_open = env.render("human")

            if ctrl.reset == 1:
                ctrl.reset = 0
                env.unwrapped.scene.cpp_world.clean_everything()
                break
```
Setup: Ubuntu 16.04, python3.5 and NVS 5200m. 
@olegklimov , could you verify if it is reproducible on your side? Thx. The bullet synchronization might be the reason, however the above code has this issue in every trial on my laptop.

olegklimov, 2017-08-23:
Works for me. I also print "RESET!!!" on R key, here is output:

```
RESET!!!!
[  5.02298586e-02   1.04617534e-18   8.98036584e-02   2.73448292e-17
   2.74748933e-02  -9.22705692e-20   3.77430804e-02  -2.73448292e-17
   8.74153897e-02   3.80974497e-19   5.04383072e-02  -6.38046030e-17
  -6.91929087e-02  -6.15552036e-19   2.33377814e-02   8.20344858e-17]
[  5.02298586e-02   2.07956577e-18   8.98036584e-02   2.73448292e-17
   2.74748933e-02   7.58196024e-19   3.77430804e-02  -5.46896583e-17
   8.74153897e-02   1.24930773e-18   5.04383072e-02  -1.54954037e-16
  -6.91929087e-02  -6.99489103e-19   2.33377814e-02   1.36724154e-16]
[  5.02298586e-02   2.47618667e-18   8.98036584e-02   1.82298861e-17
   2.74748933e-02   1.58992179e-18   3.77430804e-02  -1.09379317e-16
   8.74153897e-02   2.69867199e-18   5.04383072e-02  -2.09643698e-16
  -6.91929087e-02  -2.23215852e-19   2.33377814e-02   1.82298868e-16]
[  5.02298586e-02   2.41469923e-18   8.98036584e-02   2.73448292e-17
   2.74748933e-02   1.59473060e-18   3.77430804e-02  -1.36724154e-16
   8.74153897e-02   3.33007467e-18   5.04383072e-02  -2.27873568e-16
  -6.91929087e-02  -9.81635189e-19   2.33377814e-02   2.36988516e-16]
[  5.02298586e-02   2.34616670e-18   8.98036584e-02   5.46896583e-17
   2.74748933e-02   1.96172913e-18   3.77430804e-02  -1.82298868e-16
   8.74153897e-02   3.37116177e-18   5.04383072e-02  -3.00793126e-16
  -6.91929087e-02  -1.98917126e-18   2.33377814e-02   2.82563230e-16]
[  5.02298586e-02   2.62382621e-18   8.98036584e-02   1.27609206e-16
   2.74748933e-02   2.94594626e-18   3.77430804e-02  -2.73448308e-16
   8.74153897e-02   4.15538813e-18   5.04383072e-02  -3.55482787e-16
  -6.91929087e-02  -2.36011689e-18   2.33377814e-02   3.55482787e-16]
[  5.02298586e-02   2.49597345e-18   8.98036584e-02   1.73183920e-16
   2.74748933e-02   4.11564334e-18   3.77430804e-02  -3.73712683e-16
   8.74153897e-02   4.14655095e-18   5.04383072e-02  -4.55747136e-16
  -6.91929087e-02  -4.19779189e-18   2.33377814e-02   4.46632215e-16]
[  5.02298586e-02   2.06398748e-18   8.98036584e-02   1.91413802e-16
   2.74748933e-02   5.01021849e-18   3.77430804e-02  -4.55747136e-16
   8.74153897e-02   4.17469659e-18   5.04383072e-02  -5.19551746e-16
  -6.91929087e-02  -4.75981852e-18   2.33377814e-02   4.92206928e-16]
[  5.02298586e-02   3.64529107e-18   8.98036584e-02   1.91413802e-16
   2.74748933e-02   4.71148139e-18   3.77430804e-02  -4.92206928e-16
   8.74153897e-02   4.33696461e-18   5.04383072e-02  -5.56011538e-16
  -6.91929087e-02  -5.91595438e-18   2.33377814e-02   5.28666720e-16]
[  5.02298586e-02   5.00425121e-18   8.98036584e-02   2.27873568e-16
   2.74748933e-02   5.92738726e-18   3.77430804e-02  -5.65126459e-16
   8.74153897e-02   5.00698050e-18   5.04383072e-02  -6.19816147e-16
  -6.91929087e-02  -6.78514874e-18   2.33377814e-02   5.92471330e-16]
```

Visibly nothing is wrong with it.

navigator8972, 2017-09-08:
The above code and my baxter example work fine if bullet is initialized through ```b3ConnectSharedMemory(SHARED_MEMORY_KEY)```, instead of ```b3ConnectPhysicsDirect()``` in ```physics-bullet.cpp```.

In order to use this, we need to launch a standalone server. Run ```./build_cmake_pybullet_double.sh``` under bullet3 to build the shared memory examples (located in build_cmake, will not interfere the build for roboschool). Then run ```$BULLET3_PATH/build_cmake/examples/SharedMemory/App_PhysicsServer_SharedMemory_GUI``` before starting the roboschool example.

However, I don't quite understand this because for both ```PhysicsDirect``` and ```SharedMemory```, the reset command seems to be ultimately processed by ```PhysicsServerCommandProcessor::resetSimulation()```. And I think it makes more sense for roboschool to use ```PhysicsDirect``` as a standalone simulator.

@olegklimov do you have any idea about the readiness of current ```PhysicsDirect```? I noticed it is commented out in your and current bullet3 branches. I tried to uncomment it and test it with ExampleBrowser but none of the buttons worked (create cylinder/cubic, etc.).