openai/roboschool #56: gym_forward_walker.py:50: RuntimeWarning: invalid value encountered in greater

jasjisdo, 2017-08-24:
I got the following **RuntimeWarning** after run the `RoboschoolHumanoidFlagrun_v0_2017may.py`for serveral minutes:

```
/Users/domann/miniconda2/envs/openai/roboschool/roboschool/gym_forward_walker.py:50: RuntimeWarning: invalid value encountered in greater
  self.joints_at_limit = np.count_nonzero(np.abs(j[0::2]) > 0.99)
~INF~ [ nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan
  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan
  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan  nan   0.   0.]
score=nan in 1 frames
Traceback (most recent call last):
  File "/Users/domann/miniconda2/envs/openai/roboschool//agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 492, in <module>
    demo_run()
  File "/Users/domann/miniconda2/envs/openai/roboschool//agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py", line 34, in demo_run
    obs, r, done, _ = env.step(a)
  File "/Users/domann/miniconda2/envs/openai/roboschool/gym/gym/core.py", line 96, in step
    return self._step(action)
  File "/Users/domann/miniconda2/envs/openai/roboschool/gym/gym/wrappers/time_limit.py", line 36, in _step
    observation, reward, done, info = self.env.step(action)
  File "/Users/domann/miniconda2/envs/openai/roboschool/gym/gym/core.py", line 96, in step
    return self._step(action)
  File "/Users/domann/miniconda2/envs/openai/roboschool/roboschool/gym_forward_walker.py", line 91, in _step
    self.apply_action(a)
  File "/Users/domann/miniconda2/envs/openai/roboschool/roboschool/gym_mujoco_walkers.py", line 110, in apply_action
    assert( np.isfinite(a).all() )
```

Packages install in my env:

```
#
certifi                   2017.7.27.1               <pip>
chardet                   3.0.4                     <pip>
icu                       54.1                          0
idna                      2.6                       <pip>
mkl                       2017.0.3                      0
numpy                     1.13.1                    <pip>
openssl                   1.0.2l                        0
pip                       9.0.1                    py36_1
pyglet                    1.2.4                     <pip>
python                    3.6.2                         0
qt                        5.6.2                         2
readline                  6.2                           2
requests                  2.18.4                    <pip>
setuptools                27.2.0                   py36_0
six                       1.10.0                    <pip>
sqlite                    3.13.0                        0
tk                        8.5.18                        0
urllib3                   1.22                      <pip>
wheel                     0.29.0                   py36_0
xz                        5.2.3                         0
zlib                      1.2.11                        0
```
should I use a older numpy version? or python 3.5.x?

Maybe the code should check for `nan` in in `j`. (just a wild guess)

thanks for help with python version chaos ;)

olegklimov, 2017-08-24:
>     assert( np.isfinite(a).all() )

It does.

That crash never happens to me. I also didn't hear it happened to anybody. For now it's likely a random problem with your setup.

Let me know if you find a cause.

closed 2017-08-24 by olegklimov

gaoyuankidult, 2018-02-07:
@jasjisdo 
Hi, I wonder did you solve the problem yet? I also face the same problem.

DianaKessler, 2018-02-12:
Hello,

I am also having this problem on one of my pcs. I also posted a new issue with it a month ago, but did no get any answer until now.