openai/roboschool #58: Install problems due to missing packages

zacwellmer, 2017-09-03:
After attempting to install roboschool in the final step I'm receiving a few errors that look like the following:
```bash
Package Qt5Widgets was not found in the pkg-config search path.
Perhaps you should add the directory containing `Qt5Widgets.pc'
to the PKG_CONFIG_PATH environment variable
No package 'Qt5Widgets' found
```

I'm not sure where I'm supposed to be grabbing these .pc files from.

locke-phan, 2017-09-03:
@zacwellmer what platform are you trying to compile it for? did the compilation process succeed?

panda1221, 2017-09-21:
Hi all,
I have the same issue.
I use Mac os to build it.
Who can help on this issue?

Thank you

olegklimov, 2017-09-21:
> supposed to be grabbing these .pc files from.

`PKG_CONFIG_PATH` does that. Please see if instructions in README help. Check the .pc files are at location that `PKG_CONFIG_PATH` points to.

closed 2017-11-11 by zacwellmer