openai/roboschool #60: Different reward to MuJoCo environment

zhangwj618, 2017-09-04:
Hi,
I'm trying to use the 'RoboschoolReacher-v1' environment, and spotted difference in the ways rewards are calculated compared to the 'Reacher-v1' environment in Gym based on MuJoCo. Why is there such a difference? If I want to replicate the results by others using MuJoCo-Reacher (as in the [PPO](https://arxiv.org/abs/1707.06347) paper and [ACKTR & A2C](https://arxiv.org/abs/1708.05144) paper), can I modify the lines of code for calculating reward to be the same as the MuJoCo environment and expect the same behavior from the Roboschool environment? Thanks in advance!

olegklimov, 2017-09-04:
Hi. Please read the the blog post, link is in README.

closed 2017-09-04 by olegklimov

aznshodan, 2018-07-30:
Why is this closed? Can you guys make it so that both RoboschoolReacher and Mujoco compute rewards the same way?