openai/roboschool #61: How to train to get similar results to pre-trained demos?

erwincoumans, 2017-09-06:
The pretrained agents look very nice. Is the software used to train them available and if so, are there some instructions, training scripts?

olegklimov, 2017-09-06:
That's a bit complicated. The code released in baselines is definitely good enough for humanoid to walk, maybe more. It's just three month old simplified version of the code we had there. So it definitely works for all envs there, maybe except most complex Harder and Atlas tasks. We moved on since then, our current code is brilliant of course, but super-messy and no way near a public release )

erwincoumans, 2017-09-06:
Hah, thanks for the explanation. Hope OpenAI will spill those beans eventually, no matter how messy.

olegklimov, 2017-09-06:
By simplified code, I mean dependencies removed, we were not making the algorithm worse!

closed 2017-09-07 by erwincoumans