openai/roboschool #63: feet_collision_cost always zero

erwincoumans, 2017-09-09:
In gym_forward_walker.py there is this code:

feet_collision_cost = 0.0
        for i,f in enumerate(self.feet):
            contact_names = set(x.name for x in f.contact_list())
            print("CONTACT OF '%s' WITH %s" % (f.name, ",".join(contact_names)) )
            self.feet_contact[i] = 1.0 if (self.foot_ground_object_names & contact_names) else 0.0
            if contact_names - self.foot_ground_object_names:
                feet_collision_cost += self.foot_collision_cost

It seems to be doing an odd if statement, subtracting strings, and it never goes there. Is that intentional?

olegklimov, 2017-09-15:
Hi Erwin! If statement checks for the set of contact objects intersecting with the set of ground objects, by name.
Subtraction is on sets, meaning contacts that is not ground.

olegklimov, 2017-09-15:
(if you're implementing roboschool-like environments in pybullet, I'd recommend to do that in C++, because the overhead of using python for low level code is very high. IIRC, 50% of time spent was not simulation.)

erwincoumans, 2017-09-15:
Thanks for the explanation, that makes sense now.

>> the overhead of using python

Yes. First I'd like to have feature parity with Roboschool on all environments, disabling random and then given an action, getting the same reward and observations. Once that is there, I'd like to do the same for the existing (easier) 10 Gym Mujoco tasks. Then, optimizations can start, indeed using C++ and possibly vectorization.

closed 2017-09-15 by erwincoumans