openai/roboschool #64: Strange problem with path?

laksefest, 2017-09-13:
I think I finished installing everything, but I must have done something wrong. Maybe something very simple and Im too stupid to understand what it is. Could anyone help? 

If I try this in the roboschool main directory:
(py36) roboschool@test:~/roboschool$ python
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import roboschool
>>> 

Seems to work! Roboschool is loaded with no error. But then I change the directory:

(py36) roboschool@test:~/roboschool$ cd agent_zoo/
(py36) roboschool@test:~/roboschool/agent_zoo$ python
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import roboschool
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'roboschool'
>>> 

Why cant I load roboschool when I change my directory? Is this why the examples are not working? Seems very strange. 

(py36) roboschool@test:~/roboschool$ python $ROBOSCHOOL_PATH/agent_zoo/demo_race2.py
Traceback (most recent call last):
  File "/home/user/roboschool/agent_zoo/demo_race2.py", line 1, in <module>
    import gym, roboschool
ModuleNotFoundError: No module named 'roboschool'
(py36) roboschool@test:~/roboschool$ 

Any idea? Im using Anaconda where i made a new environment for Python 3.6 and followed the instructions to install. The install was completed:

(py36) roboschool@test:~/roboschool/bullet3/build$ cd ../..
(py36) roboschool@test:~/roboschool$ sudo -H pip install -e $ROBOSCHOOL_PATH
[sudo] password for user: 
Obtaining file:///home/user/roboschool
Installing collected packages: roboschool
  Running setup.py develop for roboschool
Successfully installed roboschool-1.0

I have tried installing with pip and pip3. Same result. Any ideas? Would be greatly appreciated! :-)

locke-phan, 2017-09-13:
I believe anaconda stores python packages in a different location compare to pip/pip3. You could try to locate where pip install python packages and append that path to environment variable PYTHONPATH.

olegklimov, 2017-11-07:
> But then I change the directory

Right, if python sees roboschool folder in CWD, it can load it as module. But when you move out, it can't. Probably what happens is you actually don't have it installed. When it works it works from CWD.

closed 2017-11-07 by olegklimov