openai/roboschool #67: How to self define a robot and put the camera on to robot's body

k123jack, 2017-09-19:
I want to define a 7 axis manipulator and put a camera onto it's hand to do the DQN training.
How can I import the mujoco xml file and make a new environment, the register process seems complicated.
And how to define a camera to capture the end-effector image. 
https://github.com/openai/roboschool/wiki/Test-Window-and-Camera-Test-Window
I reffered to this issue, but it didn't work.
Or should I use ROS/Gazebo and integrate it with gym?
Thank you

olegklimov, 2017-09-19:
> but it didn't work.

I can confirm you can make it work in roboschool current codebase.

k123jack, 2017-09-21:
I tried to create mycam, but most of the envs don't have the  attribute 'cpp_world'.
The error is : 
mycam = env.unwrapped.scene.cpp_world.new_camera_free_float(320, 200, "my_camera")
AttributeError: 'NoneType' object has no attribute 'cpp_world'

Only the env of "demokeyboard_humanoid2" has the  attribute and can create mycam.
But the picture captured seems wrong, maybe the order of reshaping pixels from string to array by this code is not correct :
rgb_array       = np.fromstring(rgb, dtype=np.uint8).reshape( (320,200,3) )

olegklimov, 2017-09-21:
> picture captured seems wrong

Try recording video using any of the zoo policies. Does this work? If it does, picture is OK, it is used to feed ffmpeg with images.

> continue to render

Call test window each frame, so it gets control to render.

k123jack, 2017-09-22:
env.render("human") works well on any of the zoo policies, env.render("rgb_array") can be played through PygletInteractiveWindow.imshow(). But env.render("rgb_array") can not be played by testwindow, the test_window doesn't continue to render, even by calling test_window each frame. I also can not import the ffmpeg module.

What I want to do is to use mycam to render, or is there anyway to change the position of camera called by env.render()?

k123jack, 2017-09-22:
For example, when I call env.env.camera.move_and_look_at(1,1,1,0,0,0), in RoboschoolReacher_v0_2017may.py before env.render(), the camera position seems doesn't change. @olegklimov

olegklimov, 2017-09-22:
Oh this doesn't work by magic, right, just look for another calls to move_and_look_at() and set a breakpoint there.

olegklimov, 2017-09-22:
closing -- nothing seems to be in a need of fixing

closed 2017-09-22 by olegklimov