openai/roboschool #68: relocation R_X86_64_32S against `.bss' can not be used when making a shared object; recompile with -fPIC

weiliu620, 2017-09-19:
I'm on CentOS 7 and Anaconda environment python 3.4, and didn't strictly follow the install instruction, which seems assume Ububntu. But here is the error when I run `pip install`: 

```
Obtaining file:///home/weiliu/packages/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
    Link against python 3.4
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
    Link against python 3.4
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-hud.cpp  -o.build-release/render-hud.o -MMD -MF .build-release/render-hud.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple-primitives.cpp  -o.build-release/render-simple-primitives.o -MMD -MF .build-release/render-simple-primitives.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.4` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c python-binding.cpp  -o.build-release/python-binding.o -MMD -MF .build-release/python-binding.o.dep
    gcc -shared -o../cpp_household.so .build-release/physics-bullet.o .build-release/assets-mesh.o .build-release/random-world-tools.o .build-release/render-glwidget.o .build-release/render-hud.o .build-release/render-simple.o .build-release/render-simple-primitives.o .build-release/python-binding.o -L/usr/lib64 -lm -lGL -lGLU -lstdc++ `pkg-config --libs Qt5OpenGL Qt5Widgets assimp python-3.4` -Wl,-z,origin -Wl,--disable-new-dtags -Wl,-rpath,'$ORIGIN/cpp-household/bullet_local_install/lib' -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API -lboost_python-py34
    /usr/bin/ld: .build-release/render-glwidget.o: relocation R_X86_64_32S against `.bss' can not be used when making a shared object; recompile with -fPIC
    .build-release/render-glwidget.o: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    make: *** [../cpp_household.so] Error 1
    cd /home/weiliu/packages/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/weiliu/packages/roboschool/
```
I did install assimp-devel, bullet, tinyxml-devel, boost-devel, and most Qt5 library (not sure about its opengl package since I coulnd't find RPM packages, maybe not needed). The error seems not related to any package dependencies, though. 

I can also find files under `roboschool/cpp-household/bullet_local_install/lib/` and `include` folder.

olegklimov, 2017-09-20:
Hardly I'll be able to help you there. Usually -fPIC means really what is says: some of the binaries compiled without -fPIC and it needs it to link a shared library. So you have to find what it is. Also check if you're building for different architectures.

weiliu620, 2017-09-20:
Thanks. But the error log above shows that `.build-release/render-glwidget.o` was compiled indeed with `-fPIC`. Did I miss anything?

olegklimov, 2017-11-07:
".bss" can indicate different architectures as well. If you have a fix that will work for more people, please post it.

closed 2017-11-07 by olegklimov

kkjh0723, 2017-11-10:
@weiliu620 Did you fix this problem? I also got the same error...

yuchen93, 2017-12-16:
I  encountered the same error. I installed assimp from source, would that be the problem?