openai/roboschool #69: Multiplayer client sending invalid string in demo_pong

DennisSoemers, 2017-09-20:
When I try running demo_pong.py, I consistently get an error after running for a little while (approximately 15 seconds I guess, definitely don't have to sit around waiting for a long time for it to happen). Here's the full output:

    $ python demo_pong.py
    Waiting /tmp/multiplayer_pongdemo_player00
    Waiting /tmp/multiplayer_pongdemo_player01
    [2017-09-20 10:54:15,046] Making new env: RoboschoolPong-v1
    Player 0 connected, wants to operate RoboschoolPong-v1 in this scene
    [2017-09-20 10:54:15,048] Making new env: RoboschoolPong-v1
    [2017-09-20 10:54:15,056] Making new env: RoboschoolPong-v1
    Player 1 connected, wants to operate RoboschoolPong-v1 in this scene
    [2017-09-20 10:54:15,058] Making new env: RoboschoolPong-v1
    4:0 Traceback (most recent call last):
      File "demo_pong.py", line 26, in <module>
        gameserver.serve_forever()
      File "/home/dennis/roboschool/roboschool/multiplayer.py", line 279, in serve_forever
        p.read_and_apply_action()
      File "/home/dennis/roboschool/roboschool/multiplayer.py", line 216, in read_and_apply_action
        raise ValueError("multiplayer client %s sent here invalid string '%s'" % (self.prefix, check))
    ValueError: multiplayer client /tmp/multiplayer_pongdemo_player00 sent here invalid string ''

Is this a known issue? Anything I can do to fix it?

The other demos (demo_race1.py and demo_race2.py) appear to work just fine, I can leave them up running for a while with no problems.

In case it's of any importance, I'm on Ubuntu 16.04.3, Python 3.6.1 (Anaconda), 64 bit.

olegklimov, 2017-09-21:
Oh, it's just 1000 frames timeout.

closed 2017-09-21 by olegklimov

DennisSoemers, 2017-09-22:
In case anyone later runs into the same problem and is curious how to fix it:

The maximum number of episodes is defined close to the bottom in this file: https://github.com/openai/roboschool/blob/master/roboschool/__init__.py

To run two-player pong for a larger number of episodes, either modify that file directly on your PC (though that may lead to difficulties when moving to different machines or reinstalling/updating roboschool for example), or simply copy/paste the relevant parts of the code somewhere else, change the max_episode_steps, and also change the id (this results in registering a new variant of the environment with a larger number of episodes, but otherwise functions exactly the same way).