openai/roboschool #71: Can I use xacro file?

k123jack, 2017-09-29:
Can I directly open a xacro file using gym_urdf_robot_env?

olegklimov, 2017-09-29:
No. Also, xacro and other ROS machinery is complicated and poorly thought out. So I don't think we need more support for it. For instance, Mujoco XMLs are shorter and better and more capable, and you don't see DLL loads there.

closed 2017-09-29 by olegklimov

k123jack, 2017-10-02:
But most of the mujoco XMLs from here :  http://www.mujoco.org/forum/index.php?resources/categories/models.3/   are not supported too.

olegklimov, 2017-10-02:
Right, this project is not a mujoco emulator.

k123jack, 2017-10-02:
So, it also doesn't support friction or etc, and we can not train grasping.

olegklimov, 2017-10-02:
Of course there is friction, otherwise robots will be falling all the time, you can even control it from .xml I believe. If your effort is something serious, you should talk to Erwin about friction and the state of it in Bullet. Pretty good according to demos you may find in Bullet.

k123jack, 2017-10-03:
Thanks,  I really need to know about the state and friction and to improve the architecture of the network.