openai/roboschool #72: cpp_household compile error on macos sierra

helloiss, 2017-10-04:
I have followed the instruction to build the roboschool project. but i get error when run 
pip3 install -e $ROBOSCHOOL_PATH
I got the following errors:

typeinfo for boost::python::objects::caller_py_function_impl<boost::python::detail::caller<Pose (Thingy::*)(), boost::python::default_call_policies, boost::mpl::vector2<Pose, Thingy&> > > in python-binding.o
      ...
**ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [../cpp_household.so] Error 1**

after hours and hours google. I still can't fix this problem. anyone can help?

olegklimov, 2017-10-04:
Oh must be something trivial. Make sure you have xcode updated, only one python visible, boost-python compiled against the same python, cd cpp-household, run make there, look at errors more closely.

helloiss, 2017-10-09:
@olegklimov Thanks for your help, I have python2.7 and python3.6 environment. after alias python=python3. I have build roboschool succeed.

closed 2017-10-10 by olegklimov