openai/roboschool #73: Segmentation fault when multiple environment/camera rendering

zhangwj618, 2017-10-04:
I'm running the Reacher environment but this issue is related to custom camera and could happen in other environments too. The following piece of code can replicate my issue (segmentation fault) when run:

`import gym, roboschool

for _ in range(2):
    env = gym.make('RoboschoolReacher-v1')
    env.reset()
    mycam = env.unwrapped.scene.cpp_world.new_camera_free_float(42, 42, "my_camera")
    mycam.move_and_look_at(0, 0, 0.283, 0, 0, 0)
    rgb, _, _, _, _ = mycam.render(False, False, False)`

If you unroll the loop and name the two envs differently, you can actually see a more specific error message:

> python3: /usr/include/boost/smart_ptr/shared_ptr.hpp:710: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = SimpleRender::Context; typename boost::detail::sp_member_access<T>::type = SimpleRender::Context*]: Assertion `px != 0' failed.

olegklimov, 2017-10-04:
Line number is one context higher on track trace.

zhangwj618, 2017-10-06:
I hope this is more useful:
> (gdb) backtrace
> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> #1  0x00007ffff6c68f5d in __GI_abort () at abort.c:90
> #2  0x00007ffff6c5ef17 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x7fffecfa1f71 "px != 0", 
>     file=file@entry=0x7fffecfa1db8 "/usr/include/boost/smart_ptr/shared_ptr.hpp", line=line@entry=710, 
>     function=function@entry=0x7fffecfa44c0 <boost::shared_ptr<SimpleRender::Context>::operator->() const::__PRETTY_FUNCTION__> "typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = SimpleRender::Context; typename boost::detail::sp_member_access<T>::type = SimpleRender::Context*]") at assert.c:92
> #3  0x00007ffff6c5efc2 in __GI___assert_fail (assertion=assertion@entry=0x7fffecfa1f71 "px != 0", 
>     file=file@entry=0x7fffecfa1db8 "/usr/include/boost/smart_ptr/shared_ptr.hpp", line=line@entry=710, 
>     function=function@entry=0x7fffecfa44c0 <boost::shared_ptr<SimpleRender::Context>::operator->() const::__PRETTY_FUNCTION__> "typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = SimpleRender::Context; typename boost::detail::sp_member_access<T>::type = SimpleRender::Context*]") at assert.c:101
> #4  0x00007fffecf55607 in boost::shared_ptr<SimpleRender::Context>::operator-> (this=<optimized out>) at /usr/include/boost/smart_ptr/shared_ptr.hpp:710
> #5  0x00007fffecf6e9b8 in boost::shared_ptr<SimpleRender::Context>::operator-> (this=0x23f2710) at /usr/include/boost/smart_ptr/shared_ptr.hpp:672
> #6  Household::Camera::camera_render (this=<optimized out>, cx=..., render_depth=render_depth@entry=false, render_labeling=render_labeling@entry=false, 
>     print_timing=<optimized out>) at render-glwidget.cpp:38
> #7  0x00007fffecf9b9cf in Camera::render (this=0x7fffe2306100, render_depth=false, render_labeling=false, print_timing=<optimized out>)
>     at python-binding.cpp:230
> #8  0x00007fffecf8cc95 in boost::python::detail::invoke<boost::python::to_python_value<boost::python::api::object const&>, boost::python::api::object (Camera::*)(bool, bool, bool), boost::python::arg_from_python<Camera&>, boost::python::arg_from_python<bool>, boost::python::arg_from_python<bool>, boost::python::arg_from_python<bool> > (ac2=..., ac1=..., ac0=..., tc=<synthetic pointer>..., f=<optimized out>, rc=...) at /usr/include/boost/python/detail/invoke.hpp:88
> #9  boost::python::detail::caller_arity<4u>::impl<boost::python::api::object (Camera::*)(bool, bool, bool), boost::python::default_call_policies, boost::mpl::vector5<boost::python::api::object, Camera&, bool, bool, bool> >::operator() (args_=<optimized out>, this=<optimized out>)
>     at /usr/include/boost/python/detail/caller.hpp:218
> #10 boost::python::objects::caller_py_function_impl<boost::python::detail::caller<boost::python::api::object (Camera::*)(bool, bool, bool), boost::python::default_call_policies, boost::mpl::vector5<boost::python::api::object, Camera&, bool, bool, bool> > >::operator() (this=<optimized out>, args=<optimized out>, 
>     kw=<optimized out>) at /usr/include/boost/python/object/py_function.hpp:38
> #11 0x00007fffe951fccd in boost::python::objects::function::call(_object*, _object*) const () from /usr/lib/x86_64-linux-gnu/libboost_python-py36.so.1.62.0
> #12 0x00007fffe951fec8 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_python-py36.so.1.62.0
> #13 0x00007fffe9526933 in boost::python::handle_exception_impl(boost::function0<void>) () from /usr/lib/x86_64-linux-gnu/libboost_python-py36.so.1.62.0
> #14 0x00007fffe951d2e9 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_python-py36.so.1.62.0
> #15 0x0000000000459893 in _PyObject_FastCallDict ()
>

zhangwj618, 2017-10-07:
I've submitted a pull request #74 containing a working fix for me.

closed 2017-10-10 by olegklimov