openai/roboschool #78: Suggested Environment

ericmock, 2017-10-19:
Sorry if this is the wrong forum for suggesting environment but as a long-time mechanical engineering professor I think a great environment would be training a robot to swing (on a swing).  There's a great paper from 1998 (http://www.math.cornell.edu/~rand/randpdf/swing.pdf) about this.  The mechanics and models are fairly simple (2D) but also deceivingly complex.  For example, if swinger does not touch the surroundings how does he keep going if there are energy losses (friction, air resistance, etc.)?  How is energy transferred into the system?

I am not (at all) at the stage to build this environment but would be willing to give it a try if there's interest.

Eric

olegklimov, 2017-11-07:
Hi. Most environments here are actually too easy. There is shortage of complex and thought out environments, that is complex but you still can learn it. But maybe not simple ones. Really, all contemporary RL algorithms eat 2d walkers for breakfast.

closed 2017-11-07 by olegklimov

ericmock, 2017-11-10:
I thought the purpose of the RoboSchool was education and I was just suggesting a simple start to an educational environment that could be extended readily to a chaotic system that is probably uncontrollable.  Just add a few degrees of freedom and you have yourself a real mess.

If you want hard 3D problems, how about training a robotic cat to always land on its feet?

olegklimov, 2017-11-10:
Dog. I have a dog model somewhere. But some unsolved problems stand in a way, like language and capacity allocation.

ericmock, 2017-11-10:
Never seen a dog perform that feat ;-)  Nor have I ever seen someone try to train a dog (or a human) to do that.

There's some literature on the subject of cats.  If I remember correctly you don't need too many degrees of freedom.  One could even theoretically test the hypothesis that cats without tails have a more difficult time righting themselves.