openai/roboschool #79: "Segmentation Fault: 11" when importing roboschool (macOS Sierra 10.12.6)

bchasnov, 2017-10-19:
After following the installation instructions for roboschool, importing roboschool causes python to crash with a segmentation fault:

```
$ python -c "import roboschool"
Segmentation fault: 11
```

So I made a file  `$ echo "import roboschool" > test.py`  and ran it through gdb 

```
$ gdb python
...
GNU gdb (GDB) 8.0.1
This GDB was configured as "x86_64-apple-darwin16.7.0".
...
(gdb) run test.py
Starting program: /Users/ben/anaconda3/bin/python test.py
[New Thread 0x1403 of process 825]
warning: unhandled dyld version (15)
[New Thread 0x110f of process 825]
[New Thread 0x1503 of process 825]
[New Thread 0x1603 of process 825]

Thread 2 received signal SIGSEGV, Segmentation fault.
0x0000000100071dd7 in insertdict ()
(gdb) quit
```

Here's my machine information if it's useful:
```
$ uname -a
Darwin Bens-MacBook-Pro.local 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64
$ python -V
Python 3.5.4 :: Anaconda, Inc.
$ gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
$ cmake --version
cmake version 3.9.4
$ brew list --versions | grep boost
boost 1.65.1
boost-python 1.65.1
```

bchasnov, 2017-10-19:
Looks like the seg fault can happen in two places randomly, `visit_decref()` and `insertdict()`:
```
Thread 2 received signal SIGSEGV, Segmentation fault.
0x00000001001c447d in visit_decref ()
```
Here's the gdb backtrace for the `insertdict()` failure
```
(gdb) backtrace
#0  0x0000000100071dd7 in insertdict ()
#1  0x0000000100070b7a in PyDict_Merge ()
#2  0x000000010007347f in PyDict_Copy ()
#3  0x000000010017d2ec in _PyImport_FixupExtensionObject ()
#4  0x00000001001808e6 in _PyImport_LoadDynamicModuleWithSpec ()
#5  0x000000010017fe4c in _imp_create_dynamic ()
#6  0x0000000100087c0e in PyCFunction_Call ()
#7  0x000000010014f963 in PyEval_EvalFrameEx ()
> #8  0x0000000100149be4 in _PyEval_EvalCodeWithName ()
> #9  0x000000010014ec52 in PyEval_EvalFrameEx ()
> #10 0x000000010014e990 in PyEval_EvalFrameEx ()
...
```
and for the `visit_decref()` failure
```
#0  0x00000001001c447d in visit_decref ()
#1  0x000000010006f72b in dict_traverse ()
#2  0x00000001001c32e3 in collect ()
#3  0x00000001001c4a06 in _PyObject_GC_Alloc ()
#4  0x00000001000a2245 in PyTuple_New ()
#5  0x000000010018ca96 in r_object ()
...
#12 0x000000010018cdc9 in r_object ()
#13 0x000000010018f975 in marshal_loads ()
#14 0x0000000100087c0e in PyCFunction_Call ()
#15 0x000000010014e898 in PyEval_EvalFrameEx ()
> #16 0x0000000100149be4 in _PyEval_EvalCodeWithName ()
> #17 0x000000010014ec52 in PyEval_EvalFrameEx ()
> #18 0x000000010014e990 in PyEval_EvalFrameEx ()
```

Marked lines `>` overlap for both failure points

bchasnov, 2017-10-19:
I downgraded my python from `3.5.4` -> `3.5.2` and it works now.
```
conda install python=3.5.2
```

closed 2017-10-19 by bchasnov

bnjasim, 2017-11-28:
I am getting the same error in python 3.6.3 on a mac. Any insights why it happens?

bnjasim, 2017-11-29:
Ok. I solved it by uninstalling the conda qt. I used the Qt installed by brew instead. 
Had to set the path to the python-boost in the Makfile inside roboschool/cpp-household as `BOOST_PYTHON = -L/usr/local/Cellar/boost-python/1.65.1/lib  -lboost_python3`

backpropper, 2018-02-10:
For python 3.6 on mac, can you post your Makefile? Still having troubles.

bnjasim, 2018-02-11:
@backpropper I have made short writeup earlier. http://cslab.org/blog/roboschool. Hope it is useful.