openai/roboschool #80: Command "python setup.py egg_info" failed with error code 1 in

Joqquin, 2017-10-23:
**Trying to install roboschool im getting this error, its probably very dumb, im new to using terminal and I dont really understand how it works yet**

Heres my code

Last login: Sun Oct 22 22:29:30 on ttys000
MacBook-Pro-de-Joaquin:~ joaquinjacubowicz$ git clone https://github.com/openai/roboschool.git
Cloning into 'roboschool'...
remote: Counting objects: 504, done.
remote: Total 504 (delta 0), reused 0 (delta 0), pack-reused 504
Receiving objects: 100% (504/504), 12.84 MiB | 2.75 MiB/s, done.
Resolving deltas: 100% (202/202), done.
MacBook-Pro-de-Joaquin:~ joaquinjacubowicz$ cd roboschool
MacBook-Pro-de-Joaquin:roboschool joaquinjacubowicz$ ROBOSCHOOL_PATH=/Users/joaquinjacubowicz/roboschool
MacBook-Pro-de-Joaquin:roboschool joaquinjacubowicz$ brew install python3
Updating Homebrew...
==> Auto-updated Homebrew!
Updated Homebrew from b4e1e39d to 357b7bb0.
Updated 1 tap (homebrew/core).
==> New Formulae
asciidoctorj               laszip@2                   sc-im
git-appraise               lgogdownloader             swift-protobuf
glib-openssl               libbitcoin-database        traefik
kontena                    postgresql@9.6
==> Updated Formulae
boost ✔                    git-annex                  open-scene-graph
boost-python ✔             git-lfs                    opencv@2
gmp ✔                      git-quick-stats            osm2pgrouting
abcmidi                    gitlab-runner              pacapt
ammonite-repl              glassfish                  paket
angular-cli                gnutls                     passenger
ansible-lint               goenv                      pgroonga
appscale-tools             gofabric8                  pgrouting
arangodb                   google-java-format         plantuml
aria2                      gopass                     pldebugger
armadillo                  grpc                       ponyc
atk                        gstreamermm                postgis
aws-sdk-cpp                gtk-gnutella               postgresql
bash-preexec               h2o                        presto
bazel                      heroku                     radare2
bear                       hugo                       ripgrep
bit                        idnits                     root
bitcoin                    irssi                      ruby
bro                        itstool                    ruby@1.8
caf                        ivykis                     ruby@1.9
camlp5                     jack                       ruby@2.0
check                      jhipster                   ruby@2.1
citus                      languagetool               ruby@2.2
cmark                      leiningen                  ruby@2.3
cockroach                  libbitcoin-blockchain      s-nail
consul                     libbitcoin-node            s3cmd
convox                     libbitcoin-server          scala
coq                        libcouchbase               scalaenv
cpprestsdk                 libgphoto2                 schismtracker
crystal-icr                libgraphqlparser           sdl_gfx
cython                     liblcf                     sonarqube
dcm2niix                   libmatio                   sourcekitten
deja-gnu                   libphonenumber             speedtest-cli
dep                        librest                    stanford-parser
dependency-check           librsync                   stubby
digdag                     libwebsockets              suricata
django-completion          lxc                        swiftformat
docker                     macvim                     swiftgen
docker-completion          mailutils                  teleport
docker-machine             mawk                       temporal_tables
docker-machine-completion  megacmd                    terraform-docs
doitlive                   memcached                  terragrunt
dovecot                    mkdocs                     thefuck
duck                       moc                        ttyd
dvm                        molecule                   urh
e2fsprogs                  mosquitto                  vert.x
[... 866 lines cut ...]
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-hud.cpp  -o.build-release/render-hud.o -MMD -MF .build-release/render-hud.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple.cpp  -o.build-release/render-simple.o -MMD -MF .build-release/render-simple.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-simple-primitives.cpp  -o.build-release/render-simple-primitives.o -MMD -MF .build-release/render-simple-primitives.o.dep
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c python-binding.cpp  -o.build-release/python-binding.o -MMD -MF .build-release/python-binding.o.dep
    gcc -shared -o../cpp_household.so .build-release/physics-bullet.o .build-release/assets-mesh.o .build-release/random-world-tools.o .build-release/render-glwidget.o .build-release/render-hud.o .build-release/render-simple.o .build-release/render-simple-primitives.o .build-release/python-binding.o -framework OpenGL -lstdc++ `pkg-config --libs Qt5OpenGL Qt5Widgets assimp python-3.6`  -Lbullet_local_install/lib -lLinearMath -lBullet3Common -lBulletCollision -lBulletDynamics -lBulletInverseDynamics -lPhysicsClientC_API -lboost_python3
    ld: library not found for -lboost_python3
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make: *** [../cpp_household.so] Error 1
    cd /Users/joaquinjacubowicz/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /Users/joaquinjacubowicz/roboschool/
MacBook-Pro-de-Joaquin:build joaquinjacubowicz$

VerseChow, 2017-11-01:
I also have this issue and don't know how to solve that. My output is like:
```
Complete output from command python setup.py egg_info:
    found resource files: 177
    running egg_info
    make: *** No rule to make target `clean'.  Stop.
    cd /home/verse/Documents/eecs498/hw3/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/verse/Documents/eecs498/hw3/roboschool/
```

abhiramm7, 2017-11-07:
I had the same error, this fixed it
```
brew install pkg-config
pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6
pip3 install -e $ROBOSCHOOL_PATH
```

VerseChow, 2017-11-07:
I think it may require python 3.5, after I refresh my laptop to ubuntu 16 and use python 3.5, it works.

olegklimov, 2017-11-07:
There are a lot of ways things can go wrong. Compiling this project has proven hard to some users, that's something to learn from, but it's possible. If you just take clean ubuntu and follow instructions, it works. If someone finds a fix that will help more people, please post it.

closed 2017-11-07 by olegklimov