openai/roboschool #87: Problem with `pip3 install -e $ROBOSCHOOL_PATH`

VerseChow, 2017-11-01:
The following is my output after I run the command. Looks like the problem is `cd /home/verse/Documents/eecs498/hw3/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so` I only have `bullet_local_install` in `cpp-household` folder, no make file found.
```
    Complete output from command python setup.py egg_info:
    found resource files: 177
    running egg_info
    make: *** No rule to make target `clean'.  Stop.
    cd /home/verse/Documents/eecs498/hw3/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in 
```

Kairobo, 2017-11-04:
Please put the bullet under roboschool's folder to install.

olegklimov, 2017-11-07:
I guess this problem is trivial. Let me know if there's something to fix.

closed 2017-11-07 by olegklimov