openai/roboschool #88: Issue installing via 'pip3 install -e $ROBOSCHOOL_PATH'

robertmacyiii, 2017-11-04:
After following the steps leading up to 'pip3 install -e $ROBOSCHOOL_PATH' I get this output on the final command:

Obtaining file:///Users/rmacy/Desktop/EECS-598.498.RL-FA-2017-/rmacy.Assignment3/roboschool
    Complete output from command python setup.py egg_info:
    found resource files: 185
    running egg_info
    Mac, assumed python3, homebrew branch
    Link against python 3.6
    Slow hardware or software render (no shadows)
    rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
    rm -rf .generated
    rm -rf .build-debug
    rm -rf .build-release
    Mac, assumed python3, homebrew branch
    Link against python 3.6
    Slow hardware or software render (no shadows)
    mkdir -p .generated
    mkdir -p .build-release
    mkdir -p .build-debug
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/System/Library/Frameworks/OpenGL.framework/Headers `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.6` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
    Package python-3.6 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `python-3.6.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'python-3.6' found
    In file included from physics-bullet.cpp:1:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    In file included from assets-mesh.cpp:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    make: *** [.build-release/physics-bullet.o] Error 1
    make: *** Waiting for unfinished jobs....
    1 error generated.
    make: *** [.build-release/assets-mesh.o] Error 1
    In file included from random-world-tools.cpp:3:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/random-world-tools.o] Error 1
    In file included from render-glwidget.cpp:2:
    In file included from ./render-glwidget.h:1:
    ./render-simple.h:6:10: fatal error: 'QtWidgets/QOpenGLWidget' file not found
    #include <QtWidgets/QOpenGLWidget>
             ^~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make: *** [.build-release/render-glwidget.o] Error 1
    cd /Users/rmacy/Desktop/EECS-598.498.RL-FA-2017-/rmacy.Assignment3/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
    C++ dependencies for this project are:
    
    bullet
    tinyxml
    boost_python
    assimp
    Qt5
    
    If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
    Install dependencies that pkg-config cannot find.
    
    
    ----------------------------------------

I already checked the installation of these dependencies on homebrew:

Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
menhir

Warning: cmake 3.9.4_1 is already installed
Warning: tinyxml 2.6.2 is already installed
Warning: assimp 4.0.1 is already installed
Warning: ffmpeg 3.4 is already installed
Warning: qt 5.9.2 is already installed

robertmacyiii, 2017-11-04:
This problem was resolved, I had to add the path to python-3.6.pc to PKG_CONFIG_PATH.

Now I can run the next command 'python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py' but it causes a segmentation fault:

$ python $ROBOSCHOOL_PATH/agent_zoo/RoboschoolHumanoidFlagrun_v0_2017may.py
Segmentation fault: 11

olegklimov, 2017-11-07:
Good!

closed 2017-11-07 by olegklimov