openai/roboschool #92: Fail to pip install in Ubuntu 14.04, Anaconda, and python3.5

kkjh0723, 2017-11-08:
My system is Ubuntu 14.04, Anaconda and python3.5, and I got the error as below. How can I solve the problem? I read some of the previous issues but couldn't figure out the solution.

    Obtaining file:///home/jinhyung/git/roboschool
        Complete output from command python setup.py egg_info:
        found resource files: 185
        running egg_info
        Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
        Link against python 3.5
        Slow hardware or software render (no shadows)
        rm -f ../robot-test-tool ../robot-test-tool_d ../cpp_household.so ../cpp_household_d.so  .build-release/physics-bullet.o  .build-release/assets-mesh.o  .build-release/random-world-tools.o  .build-release/render-glwidget.o  .build-release/render-hud.o  .build-release/render-simple.o  .build-release/render-simple-primitives.o  .build-release/test-tool-qt4.o  .build-release/python-binding.o  .build-debug/physics-bullet.o  .build-debug/assets-mesh.o  .build-debug/random-world-tools.o  .build-debug/render-glwidget.o  .build-debug/render-hud.o  .build-debug/render-simple.o  .build-debug/render-simple-primitives.o  .build-debug/test-tool-qt4.o  .build-debug/python-binding.o .generated/*.moc *.ilk *.pdb .build-release/physics-bullet.o.dep .build-release/assets-mesh.o.dep .build-release/random-world-tools.o.dep .build-release/render-glwidget.o.dep .build-release/render-hud.o.dep .build-release/render-simple.o.dep .build-release/render-simple-primitives.o.dep .build-release/test-tool-qt4.o.dep .build-release/python-binding.o.dep .build-debug/physics-bullet.o.dep .build-debug/assets-mesh.o.dep .build-debug/random-world-tools.o.dep .build-debug/render-glwidget.o.dep .build-debug/render-hud.o.dep .build-debug/render-simple.o.dep .build-debug/render-simple-primitives.o.dep .build-debug/test-tool-qt4.o.dep .build-debug/python-binding.o.dep
        rm -rf .generated
        rm -rf .build-debug
        rm -rf .build-release
        Linux, python3, anaconda branch, will query Ubuntu python3 for boost_python
        Link against python 3.5
        Slow hardware or software render (no shadows)
        mkdir -p .generated
        mkdir -p .build-release
        mkdir -p .build-debug
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c physics-bullet.cpp  -o.build-release/physics-bullet.o -MMD -MF .build-release/physics-bullet.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC     -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c assets-mesh.cpp  -o.build-release/assets-mesh.o -MMD -MF .build-release/assets-mesh.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c random-world-tools.cpp  -o.build-release/random-world-tools.o -MMD -MF .build-release/random-world-tools.o.dep
        gcc -std=c++11 -Wall -Wno-unused-variable -Wno-unused-function -Wno-deprecated-register -fPIC -DBT_USE_DOUBLE_PRECISION -g -O3 -march=native -I/usr/include `pkg-config --cflags Qt5Widgets Qt5OpenGL assimp python-3.5` -Ibullet_local_install/include -Ibullet_local_install/include/bullet -I/usr/local/include/bullet -c render-glwidget.cpp  -o.build-release/render-glwidget.o -MMD -MF .build-release/render-glwidget.o.dep
        In file included from render-glwidget.h:1:0,
                         from physics-bullet.cpp:1:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from assets-mesh.cpp:1:0:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from random-world-tools.cpp:3:0:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        In file included from render-glwidget.h:1:0,
                         from render-glwidget.cpp:2:
        render-simple.h:6:35: fatal error: QtWidgets/QOpenGLWidget: No such file or directory
         #include <QtWidgets/QOpenGLWidget>
                                           ^
        compilation terminated.
        make: *** [.build-release/physics-bullet.o] Error 1
        make: *** Waiting for unfinished jobs....
        make: *** [.build-release/assets-mesh.o] Error 1
        make: *** [.build-release/random-world-tools.o] Error 1
        make: *** [.build-release/render-glwidget.o] Error 1
        cd /home/jinhyung/git/roboschool/roboschool/cpp-household && make clean && make -j4 dirs  ../cpp_household.so
    
        C++ dependencies for this project are:
        
        bullet
        tinyxml
        boost_python
        assimp
        Qt5
        
        If you see compilation error FIRST THING TO CHECK if pkg-config call was successful.
        Install dependencies that pkg-config cannot find.
        
    
        ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /home/jinhyung/git/roboschool/

Following is `qmake -v` output.

    QMake version 3.0
    Using Qt version 5.6.2 in /home/jinhyung/anaconda3/envs/py35/lib

And `PKG_CONFIG_PATH` is
    
    /home/jinhyung/anaconda3/envs/py35/lib/pkgconfig

closed 2017-11-08 by kkjh0723

syllogismos, 2017-11-25:
I'm having the same issue, can you tell me how you installed it?

syllogismos, 2017-11-25:
I was able to solve this issue by doing this https://github.com/openai/roboschool/issues/94

`export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:<YOUR-PYTHON-PATH>/lib/pkgconfig by replacing <YOUR-PYTHON-PATH>`

in my case as I use anaconda, I have to do this `export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/home/ubuntu/anaconda3/lib/pkgconfig`