openai/roboschool #95: libboost_python error.

jerinphilip, 2017-11-28:
Archlinux. Had to modify the Makefile to use libboost_python3 instead of libboost_python36. But the pain doesn't seem to end there.

Any possible fixes?

```py
Python 3.6.3 (default, Oct 24 2017, 14:48:20) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import roboschool
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jerin/repo/rl-sandbox/env/lib/python3.6/site-packages/roboschool/__init__.py", line 105, in <module>
    from roboschool.gym_pendulums import RoboschoolInvertedPendulum
  File "/home/jerin/repo/rl-sandbox/env/lib/python3.6/site-packages/roboschool/gym_pendulums.py", line 1, in <module>
    from roboschool.scene_abstract import SingleRobotEmptyScene
  File "/home/jerin/repo/rl-sandbox/env/lib/python3.6/site-packages/roboschool/scene_abstract.py", line 12, in <module>
    from roboschool  import cpp_household   as cpp_household
ImportError: /usr/lib/libboost_python.so.1.65.1: undefined symbol: PyString_InternFromString
```

olegklimov, 2017-11-28:
Oh I saw this kind of errors, it's just mismatching python versions.

jerinphilip, 2017-11-29:
So? What's the fix? Will I have to build roboschool under some environment linking it against required versions?

If so - is there a file containing details of versions?

ThGravo, 2017-12-21:
I'm also on Arch and ran into the same problem after modifying the `Makefile`. But linking with `-lboost_python3` instead of `-lboost_python` actually fixed it for me.
Can you do a: `ls -l /usr/lib/libboost_python*`?