openai/roboschool #96: what is human data format

xingxing1, 2017-11-30:
what is meaning weights_dense1_w in RoboschoolWalder2d_vo_2017may.py,I want use my data,but i don't know data format

olegklimov, 2017-11-30:
`_w` is matrix, `_b` is bias.

closed 2017-11-30 by olegklimov

xingxing1, 2017-12-01:
I mean _w matrix data format,such as column or row,why it is 22 row 128 column,i dont understand the mean that represent,I can't find any introduce document.

olegklimov, 2017-12-01:
It's implemented using neural networks. Please read about how they work.

xingxing1, 2017-12-04:
If I want training my data using neural networks,I should know what is standard data,I have data set including joint angle,step space,angular velocity etc.i have human gate data,but I can’t connect it with matrix that you said, I don’t know what model of human model you use. Whether or not i have misconception about roboschool.